larsimmisch/pyalsaaudio #2: Constant reproducible memory leak on Python 3.x when playing audio

larsimmisch, 2015-05-08:
This bug report was contributed by Stein Magnus Jodal over on SourceForge:

https://sourceforge.net/p/pyalsaaudio/bugs/16/

The following script consistently reproduces a memory leak on Python 3.x, but not Python 2.7. I'm running this on an amd64 system running Ubuntu 14.04.
This bug report is the result of the debugging done in https://github.com/mopidy/pyspotify/issues/127.

```
#!/usr/bin/python

from __future__ import print_function

import resource
import time

import alsaaudio

seconds = 0
max_rss = 0
device = alsaaudio.PCM()

while True:
    device.write(b'\x00' * 44100)
    time.sleep(1)
    seconds += 1
    if seconds % 10 == 0:
        prev_rss = max_rss
        max_rss = resource.getrusage(resource.RUSAGE_SELF).ru_maxrss
        diff_rss = max_rss - prev_rss
        print('After %ds: max RSS %d kB, increased %d kB' % (
            seconds, max_rss, diff_rss))
```
Results from running the above test script:
```
$ python --version
Python 2.7.6
$ python pyalsaaudio-memory-leak.py 
After 10s: max RSS 6740 kB, increased 6740 kB
After 20s: max RSS 6740 kB, increased 0 kB
After 30s: max RSS 6740 kB, increased 0 kB
After 40s: max RSS 6740 kB, increased 0 kB
After 50s: max RSS 6740 kB, increased 0 kB
After 60s: max RSS 6740 kB, increased 0 kB
After 70s: max RSS 6740 kB, increased 0 kB
After 80s: max RSS 6740 kB, increased 0 kB
After 90s: max RSS 6740 kB, increased 0 kB
After 100s: max RSS 6740 kB, increased 0 kB
After 110s: max RSS 6740 kB, increased 0 kB
After 120s: max RSS 6740 kB, increased 0 kB
After 130s: max RSS 6740 kB, increased 0 kB
After 140s: max RSS 6740 kB, increased 0 kB
After 150s: max RSS 6740 kB, increased 0 kB
After 160s: max RSS 6740 kB, increased 0 kB
After 170s: max RSS 6740 kB, increased 0 kB
After 180s: max RSS 6740 kB, increased 0 kB
```
```
$ python --version
Python 3.2.5
$ python pyalsaaudio-memory-leak.py 
After 10s: max RSS 9604 kB, increased 9604 kB
After 20s: max RSS 9904 kB, increased 300 kB
After 30s: max RSS 10432 kB, increased 528 kB
After 40s: max RSS 10696 kB, increased 264 kB
After 50s: max RSS 11224 kB, increased 528 kB
After 60s: max RSS 11644 kB, increased 420 kB
After 70s: max RSS 12172 kB, increased 528 kB
After 80s: max RSS 12436 kB, increased 264 kB
After 90s: max RSS 12964 kB, increased 528 kB
After 100s: max RSS 13492 kB, increased 528 kB
After 110s: max RSS 13756 kB, increased 264 kB
After 120s: max RSS 14284 kB, increased 528 kB
After 130s: max RSS 14812 kB, increased 528 kB
After 140s: max RSS 15076 kB, increased 264 kB
After 150s: max RSS 15604 kB, increased 528 kB
After 160s: max RSS 15868 kB, increased 264 kB
After 170s: max RSS 16396 kB, increased 528 kB
After 180s: max RSS 16924 kB, increased 528 kB
```
```
$ python --version
Python 3.4.0
$ python pyalsaaudio-memory-leak.py 
After 10s: max RSS 8252 kB, increased 8252 kB
After 20s: max RSS 8516 kB, increased 264 kB
After 30s: max RSS 9044 kB, increased 528 kB
After 40s: max RSS 9572 kB, increased 528 kB
After 50s: max RSS 9836 kB, increased 264 kB
After 60s: max RSS 10364 kB, increased 528 kB
After 70s: max RSS 10664 kB, increased 300 kB
After 80s: max RSS 11192 kB, increased 528 kB
After 90s: max RSS 11720 kB, increased 528 kB
After 100s: max RSS 11984 kB, increased 264 kB
After 110s: max RSS 12512 kB, increased 528 kB
After 120s: max RSS 13040 kB, increased 528 kB
After 130s: max RSS 13304 kB, increased 264 kB
After 140s: max RSS 13832 kB, increased 528 kB
After 150s: max RSS 14360 kB, increased 528 kB
After 160s: max RSS 14624 kB, increased 264 kB
After 170s: max RSS 15152 kB, increased 528 kB
After 180s: max RSS 15416 kB, increased 264 kB
```

The growth seems to be about (15416kB - 8252kB) / (180s - 10s) = 42.14 kB/s. This is peculiarly close to the 43.07 kB (44100 bytes / 1024) of fake audio data the test script throws at pyalsaaudio every second.
In real life, the memory growth will usually be four times higher, since 16-bit stereo audio is used instead of the 8-bit mono audio we simulate here.
Same test script, but with four times more data being written to the device:
```
$ python --version
Python 3.4.0
$ python pyalsaaudio-memory-leak.py 
After 10s: max RSS 9624 kB, increased 9624 kB
After 20s: max RSS 11208 kB, increased 1584 kB
After 30s: max RSS 13056 kB, increased 1848 kB
After 40s: max RSS 14904 kB, increased 1848 kB
After 50s: max RSS 16488 kB, increased 1584 kB
After 60s: max RSS 18336 kB, increased 1848 kB
After 70s: max RSS 20184 kB, increased 1848 kB
After 80s: max RSS 21768 kB, increased 1584 kB
After 90s: max RSS 23616 kB, increased 1848 kB
After 100s: max RSS 25464 kB, increased 1848 kB
After 110s: max RSS 27048 kB, increased 1584 kB
After 120s: max RSS 28896 kB, increased 1848 kB
After 130s: max RSS 30744 kB, increased 1848 kB
After 140s: max RSS 32328 kB, increased 1584 kB
After 150s: max RSS 34176 kB, increased 1848 kB
After 160s: max RSS 36024 kB, increased 1848 kB
After 170s: max RSS 37608 kB, increased 1584 kB
After 180s: max RSS 39456 kB, increased 1848 kB
```

larsimmisch, 2015-05-08:
Sorry it took so long

closed 2015-05-08 by larsimmisch

jodal, 2015-05-08:
A year to the day :-) Do you plan a release of pyalsaaudio in the near future?

larsimmisch, 2015-05-08:
Yes, there are still a few bugs/issues I want to fix but I think I'll do a new release next week.

commit 5e8a08bc (2015-05-08): Fix memory leak on Python 3. Closes #2