larsimmisch/pyalsaaudio #3: Cannot PIP install

a-musing-moose, 2015-05-16:
Hi @larsimmisch - I am currently unable to install on Ubuntu via pip due to an exception being raised when setup.py is run. It appears to be related to the use of git to retrieve the version number. e.g. when installing from pypi it isn't in a git repo.

```
  Running setup.py (path:/home/mossj/.virtualenvs/common-sound/build/pyalsaaudio/setup.py) egg_info for package pyalsaaudio
    fatal: Not a git repository (or any of the parent directories): .git
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/home/mossj/.virtualenvs/common-sound/build/pyalsaaudio/setup.py", line 17, in <module>
        pyalsa_version = gitrev()
      File "/home/mossj/.virtualenvs/common-sound/build/pyalsaaudio/setup.py", line 14, in gitrev
        '--always'])
      File "/usr/lib/python3.4/subprocess.py", line 616, in check_output
        raise CalledProcessError(retcode, process.args, output=output)
    subprocess.CalledProcessError: Command '['git', 'describe', '--tags', '--dirty=-dev', '--always']' returned non-zero exit status 128
    Complete output from command python setup.py egg_info:
    fatal: Not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File "<string>", line 17, in <module>
  File "/home/mossj/.virtualenvs/common-sound/build/pyalsaaudio/setup.py", line 17, in <module>
    pyalsa_version = gitrev()
  File "/home/mossj/.virtualenvs/common-sound/build/pyalsaaudio/setup.py", line 14, in gitrev
    '--always'])
  File "/usr/lib/python3.4/subprocess.py", line 616, in check_output
    raise CalledProcessError(retcode, process.args, output=output)
subprocess.CalledProcessError: Command '['git', 'describe', '--tags', '--dirty=-dev', '--always']' returned non-zero exit status 128
```

larsimmisch, 2015-05-16:
Thanks - I didn't think of that. Will fix this over the weekend.

closed 2015-05-16 by larsimmisch

a-musing-moose, 2015-05-17:
Thanks @larsimmisch! Much appreciated

commit 5cbc8860 (2015-05-16): We can get the version from git for pip installs. Fixes #3

Maybe we should tag from setup.py instead