larsimmisch/pyalsaaudio #4: test.py fails; ALSAAudioError: Device or resource busy [sysdefault:CARD=PCH]

ixbidie, 2015-06-01:
I got version 0.8.2 via [PyPi](https://pypi.python.org/pypi/pyalsaaudio)
I'm running Ubuntu 14.10 (64bit), therefore using PulseAudio
I installed libasound2-dev from dist repo (1.0.28)
I build and installed without errors

This happened when i ran test.py
<pre>
...ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave
E..ALSA lib conf.c:4705:(snd_config_expand) Unknown parameters CARD=default
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default:CARD=default
.
======================================================================
ERROR: testPCM (__main__.PCMTest)
Open a PCM object on every device
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test.py", line 96, in testPCM
    pcm = alsaaudio.PCM(device=pd)
ALSAAudioError: Device or resource busy [sysdefault:CARD=PCH]

----------------------------------------------------------------------
Ran 7 tests in 0.101s

FAILED (errors=1)
</pre>

larsimmisch, 2015-06-01:
I have seen a similar issue, also with pulseaudio. I have not investigated in detail, but when I had that problem, the failing PCM could be opened manually, like this:

```
import alsaaudio
pcm = alsaaudio.PCM('sysdefault:CARD=PCH')
```

Just in the context of the tests did that *device busy* error occur. This is a problem in the lower layers, and I think you can safely ignore that error.

anikkcah, 2016-12-11:
@larsimmisch Im facing the same problem, how to solve it.
[image]

closed 2017-02-22 by larsimmisch