larsimmisch/pyalsaaudio #8: Imprecise conversion from db to percent and vice-versa on mixer interface

hotvic, 2015-07-22:
The conversion from db to percent is inprecise.

Try increment vol from 0 to 100 by 5; the get results is: 0..5..9..14..19 and so on.

Another thing, getrange return the values range in db, while get/setvolume will get/set a percent value (0..100)

quantenschaum, 2017-09-28:
`amixer` allows to set volume in %, dB or units that are specific for the card/device. It also has the `-M` option, which allows to set the volume in %, but on a logarithmic scale, such that it seems natural. But not every soundcard supports that option. The HifiBerry-Amp+ does not.

The methods for getting and setting the volume should support all 3 kinds of units and the conversion to log scale should be done in this lib, so that it always works. Converting to log scale in the 0 to 100% range causes very noticeable loss of precision. Internally the volume should always be set using the card specific units (range).

larsimmisch, 2017-09-29:
Working around ALSA-specific quirks is not in scope for this library - it tries to give access to ALSA as it is, warts and all. Having said that, this particular workaround might be done safely. PR anyone?

quantenschaum, 2017-09-29:
This is no alsa specific quirk. To get rid of the loss of precision, pyalsaaudio shoud provide access to the soundcard specific units, the range. There is `Mixer.getrange()`, but what can you do with it? `Mixer.setvolume()` should accept these units. Accepting integer percent only makes a logarithmic volume control nearly impossible, it's too coarse. (And this is an alsa specific library, isn't it?)

### Getting the conversion right
Human perception of loudness is logarithmic, that's why [dB](https://en.wikipedia.org/wiki/Decibel) is used as unit. It's a relative unit, describing the ratio of two values.

Let's say the volume knob has the range of **[0,1]** and the [dynamic range](https://en.wikipedia.org/wiki/Dynamic_range) of the soundcard/amp is **d**. Then the knob volume has to be transformed to normalized soundcard units like this.

```python
dB = lambda x: 2*log10(x)
vol = lambda d,x: dB((1-x)/d+x)/dB(d)+1
```

For a dynamic range of **d=50** this results in. (You have know **d** or just try out, what sounds good to your ears, 50 is good start.)

[image]

Now `amixer`  or `Mixer.getrange()` tell the range of soundcard specific units **[a,b]**. This can be used to calculate absolute soundcard specific units.

```python
def volume(percent, mixer_range, dynamic_range):
    p = percent/100
    v = vol(dynamic_range, p)
    a,b = mixer_range
    return int(a*(1-v)+b*v)
```

quantenschaum, 2017-09-29:
Let's compare the percentage range of `Mixer.setvolume` **[0,100]** with soundcard specific range **[0,255]**

```python
d=50
for p in range(70,101):
    print(p, volume(p, [0,100],d), volume(p, [0,255],d))
```

outputs

```
70 91 232
71 91 233
72 91 234
73 92 234
74 92 235
75 92 236
76 93 237
77 93 238
78 93 239
79 94 239
80 94 240
81 94 241
82 95 242
83 95 243
84 95 243
85 95 244
86 96 245
87 96 246
88 96 246
89 97 247
90 97 248
91 97 248
92 97 249
93 98 250
94 98 251
95 98 251
96 98 252
97 99 253
98 99 253
99 99 254
100 100 255
```

See how the same value (2nd col) is assigned to different volume knob percentages (1st col) and how the resolution get better when using the finer grained soundcard specific range (3rd col)?

quantenschaum, 2017-09-29:
### Workaround

for volume manipulations is to `subprocess.call` `amixer`. :disappointed:

quantenschaum, 2017-09-29:
Would be nice, if you implemented something like `Mixer.setrawvolume(raw_vol)` or `Mixer.setvolume(raw_vol, raw=True)`.

quantenschaum, 2017-10-03:
Nothing?

larsimmisch, 2017-10-03:
Oh, I might actually do it. I.m thinking of buying an Amp+ anyway, so I'd have something to test with.

But don't hold your breath, I am quite busy with other things.

quantenschaum, 2020-12-22:
@larsimmisch did you find time to implement something?

May proposal is *to always use device specific units* (`amixer` raw volume units/limits). 

1. get the range of permitted values with `Mixer.getrange()`
2. use these units in `Mixer.setvolume()` (not percent). 

Conversion from/to percent or dB has to be done by the user or by *additional* helper methods provided by this library. Setting and getting the volume level to/from the device has to be done in *device specific units*, always. This way you have control over the precision in the conversion of units.

closed 2022-03-28 by larsimmisch

sixtyfive, 2025-02-05:
> dB = lambda x: 2*log10(x)
> vol = lambda d,x: dB((1-x)/d+x)/dB(d)+1

Sorry to reply to such an old Issue, but this is the best discussion of the whole subject that came up when searching. I'm trying to figure out how `alsamixer` (not `amixer`!) manages to display its volume slider controls, where the percentage of the box filled by the slider goes up and down quite linearly. At first, the two formulas quoted above seemed like they might hold the solution, but I couldn't figure it out with these either.

It still seems that knowing the raw limits (0-144 on the card I'm playing with) and the current raw value is not enough. On the other hand, alsamixer doesn't seem to be applying a per-device custom curve, but to do something that works the same with all cards. Does anyone know how it does that?

ossilator, 2025-02-06:
https://github.com/alsa-project/alsa-utils/blob/master/alsamixer/volume_mapping.c really wasn't all that hard to find ...