larsimmisch/pyalsaaudio #9: Version 0.8.2 Test failure: Invalid argument

BobBowles, 2015-07-31:
Setup and installed in the standard Debian Wheezy repository version of python3.2, including the dev package for `libasound2`.

This is the traceback of the test program:

    $ python3.2 test.py
    ...E..ALSA lib confmisc.c:768:(parse_card) cannot find card 'default'
    ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_card_driver returned error: No such device
    ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
    ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_concat returned error: No such device
    ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
    ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_refer returned error: No such device
    ALSA lib conf.c:4720:(snd_config_expand) Evaluate error: No such device
    ALSA lib pcm.c:2217:(snd_pcm_open_noupdate) Unknown PCM default:CARD=default
    .
    ======================================================================
    ERROR: testPCM (__main__.PCMTest)
    Open a PCM object on every device
    ----------------------------------------------------------------------
    Traceback (most recent call last):
      File "test.py", line 96, in testPCM
        pcm = alsaaudio.PCM(device=pd)
    alsaaudio.ALSAAudioError: Invalid argument [surround41:CARD=PCH,DEV=0]

    ----------------------------------------------------------------------
    Ran 7 tests in 0.045s

    FAILED (errors=1)

The sound card is set up correctly and can be probed by the standard alsa commands:

    $ cat /proc/asound/cards
     0 [PCH            ]: HDA-Intel - HDA Intel PCH
                          HDA Intel PCH at 0xc1610000 irq 48

    $ arecord -l
    **** List of CAPTURE Hardware Devices ****
    card 0: PCH [HDA Intel PCH], device 0: CX20590 Analog [CX20590 Analog]
      Subdevices: 1/1
      Subdevice #0: subdevice #0

I am also able to record and playback using `arecord` and `aplay`.

JohnCC330, 2015-10-01:
Basically the same problem - on a desktop machine. All other audio software (non-alsaaudio) functions normally

```
...E..ALSA lib confmisc.c:768:(parse_card) cannot find card 'default'
ALSA lib conf.c:4260:(_snd_config_evaluate) function snd_func_card_driver returned error: No such device
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4260:(_snd_config_evaluate) function snd_func_concat returned error: No such device
ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
ALSA lib conf.c:4260:(_snd_config_evaluate) function snd_func_refer returned error: No such device
ALSA lib conf.c:4739:(snd_config_expand) Evaluate error: No such device
ALSA lib pcm.c:2267:(snd_pcm_open_noupdate) Unknown PCM default:CARD=default
.
======================================================================
ERROR: testPCM (__main__.PCMTest)
Open a PCM object on every device
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test.py", line 96, in testPCM
    pcm = alsaaudio.PCM(device=pd)
ALSAAudioError: Invalid argument [surround41:CARD=SB,DEV=0]

----------------------------------------------------------------------
Ran 7 tests in 0.043s

FAILED (errors=1)
```

I suspect that this might be a missing explicit definition of a 'default' device in the alsa configuration. Will test this later.

larsimmisch, 2015-10-02:
Yes, I think this is because the alsa configuration has no default card. The tests are a little bit dumb in this regard. What a system/distribution is that?

ebezzam, 2015-10-19:
Hi Lars, 

I have a similar problem. I am running Debian GNU/Linux 7.7 (wheezy) on a Beagle Board. These are my sound cards: 

```
\root@beaglebone:~/bbb-sta321mp/pyalsaaudio-0.8.2# cat /proc/asound/cards
 0 [EVM            ]: STA321MP_EVM - STA321MP EVM
                      STA321MP EVM
 1 [Set            ]: USB-Audio - C-Media USB Headphone Set
                      C-Media USB Headphone Set at usb-musb-hdrc.1.auto-1, full speed
```

I would like to use ```hw:0,0``` for capture and ```hw:1,0``` for playback (and these worked when using ```arecord``` and ```aplay```). However, setting ```card 0``` as default as such in ```/etc/asound.conf```:

```
pcm.!default {
   type hw
   card 0
}

ctl.!default {
   type hw
   card 0
}
```

yielded the following errors:

```
root@beaglebone:~/bbb-sta321mp/pyalsaaudio-0.8.2# python test.py
.EEEEEALSA lib conf.c:4687:(snd_config_expand) Unknown parameters CARD=default
ALSA lib pcm.c:2217:(snd_pcm_open_noupdate) Unknown PCM default:CARD=default
.
======================================================================
ERROR: testMixerAll (__main__.MixerTest)
Run common Mixer methods on an open object
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test.py", line 63, in testMixerAll
    mixer = alsaaudio.Mixer(mixers[0])
IndexError: list index out of range

======================================================================
ERROR: testMixerClose (__main__.MixerTest)
Run common Mixer methods on a closed object and verify it raises an
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test.py", line 79, in testMixerClose
    mixer = alsaaudio.Mixer(mixers[0])
IndexError: list index out of range

======================================================================
ERROR: testPCM (__main__.PCMTest)
Open a PCM object on every device
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test.py", line 96, in testPCM
    pcm = alsaaudio.PCM(device=pd)
ALSAAudioError: Invalid argument [surround41:CARD=Set,DEV=0]

======================================================================
ERROR: testPCMAll (__main__.PCMTest)
Run all PCM methods on an open object
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test.py", line 106, in testPCMAll
    pcm = alsaaudio.PCM()
ALSAAudioError: No such file or directory [default]

======================================================================
ERROR: testPCMClose (__main__.PCMTest)
Run all PCM methods on a closed object and verify it raises an error
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test.py", line 120, in testPCMClose
    pcm = alsaaudio.PCM()
ALSAAudioError: No such file or directory [default]

----------------------------------------------------------------------
Ran 7 tests in 0.166s

FAILED (errors=5)
```

I also set ```card 1``` as default and got a similar error as the original poster:

```
 root@beaglebone:~/bbb-sta321mp/pyalsaaudio-0.8.2# python test.py
...E..ALSA lib conf.c:4687:(snd_config_expand) Unknown parameters CARD=default
ALSA lib pcm.c:2217:(snd_pcm_open_noupdate) Unknown PCM default:CARD=default
.
======================================================================
ERROR: testPCM (__main__.PCMTest)
Open a PCM object on every device
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test.py", line 96, in testPCM
    pcm = alsaaudio.PCM(device=pd)
ALSAAudioError: Invalid argument [surround41:CARD=Set,DEV=0]

----------------------------------------------------------------------
Ran 7 tests in 0.196s

FAILED (errors=1)
``` 

Thank you for your help.

Best,
Eric

closed 2017-02-25 by larsimmisch