larsimmisch/pyalsaaudio #10: Error in isine.py

caver-mike, 2015-09-22:
The generate(frequency) function does not properly create the buffer.  This code:

       buffer = buffer + digitize(sin(i/(2 * pi)))

Should be:

        buffer = buffer + digitize(sin((i/size)*(2 * pi)))

larsimmisch, 2015-09-22:
Your code will only generate silence, because `i` and `size` are integers and `i < size`. So `i/size` will always be zero.

caver-mike, 2015-09-22:
> Send reply to:	larsimmisch/pyalsaaudio <reply@reply.github.com>
> Copies to:	caver-mike <caveman@pemtel.net>

Hello Lars -

  I am using python3, so the result of i/size is a float.  The point 
of this code is to create a series of samples defining one cycle of 
the sine wave.  This means that the first sample should be the sine 
of zero (which is zero), and the last sample should be the sine of 
slightly less than 2*pi (very close to zero).  

When I run the isine.py module with no changes (just as I downloaded 
it), I get the message "factor: 4, frames: 400".  The code is 
supposed to generate a sine wave at 440 Hz.  Instead, I get a 
repeating pattern of about 2.5 sine waves.  The sine waves are at 
about 1000 Hz, and there is a discontinuity at the end of the buffer. 
 So, the buffer is repating at 440 Hz, as it is supposed to.  But 
since the buffer does not contain exactly one pass through the sine 
function, the output is not what is desired.

In this case, the buffer contains 100 sample values.  This means that 
the variable i iterates from 0 to 99.  Therefore, the first 
calculated sample is the sine of zero (as it should be).  However, 
the last sample value is the sine of 99/(2*pi).  This is the sine of 
15.75, which is -0.048.  In between, the sine function is using 
values that pass through 2*pi 2.5 times.

I made the change that I suggested, and now I get clean sine waves  
at the desired frequency.

I appreciate the work that you have put into this package.  My 
application is specifically to generate sine waves for automated 
testing of electronic equipment, and I prefer to work in Python.  
Thank you for providing this interface.

---- Mike Newsome

Date sent:	Tue, 22 Sep 2015 02:57:28 -0700

larsimmisch, 2015-09-22:
Thanks for your detailed answer.

My response was a bit short - your suggested solution will not work on Python 2 and I didn't have time to properly think about it.

But I'll give it some thought tomorrow and fix it for both Python2 and 3.

It's honestly a bit embarrassing I got such an elemental thing wrong.

commit 94ced051 (2017-02-25): Correct the sine example (finally!) Closes #10
closed 2017-02-25 by larsimmisch