larsimmisch/pyalsaaudio #12: Error in recordtest.py

AngeloMendes, 2015-10-02:
hello, I am trying to record an audio file with a microphone, but is giving error in the code.

  card = 'default'

Not found any directory 'default'

the variable card is my sound card?

this line: inp = alsaaudio.PCM (alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NONBLOCK, card)

I say what is the way, CAPTURE, say that is not to block, and last parameter indicates what?

I'm sorry for asking, I'm starting now with raspberry :(
And thanks for the help!

larsimmisch, 2015-10-02:
`card` is an ALSA device name - see http://larsimmisch.github.io/pyalsaaudio/libalsaaudio.html#pcm-objects.

Have a look at what `alsaaudio.pcms()` returns and pick a PCM device from there.

(I need to change the examples to be closer to the current alsaaudio version, the variable name `card` is confusing)

Also see:

http://superuser.com/questions/53957/what-do-alsa-devices-like-hw0-0-mean-how-do-i-figure-out-which-to-use

AngeloMendes, 2015-10-09:
I configured my USB Sound card as the default, but I can not play a wav using a playwav.py. If I do aplay plughw: 0.0 file.wav // it works normal. If I do playwav.py python -c 'Generic USB Audio Device' file.wav // an error occurs: 
===================== 
alsaaudio.ALSAAudioError: Invalid argument [default: CARD = Generic USB Audio Device] ===================== 
I used the aplay -L command to view the card name. If I use default to error anyway, but the USB sound card is already my default device.

AngeloMendes, 2015-10-13:
Any suggestion?

larsimmisch, 2015-10-13:
Have you tried `playwav.py -c plughw:0.0 file.wav`?

I would expect that to work.

You could also rewrite `playwav.py` to accept a `device` argument, something like:
```
if __name__ == '__main__':

    device = 'default'

    opts, args = getopt.getopt(sys.argv[1:], 'd:')
    for o, a in opts:
        if o == '-d':
            device = a

    if not args:
        usage()
        
    with wave.open(args[0], 'rb') as f:
        with alsaaudio.PCM(device=device) as adev:
            play(adev, f)
```

AngeloMendes, 2015-10-15:
Thanks, worked perfectly. Could play a file.wav with playwav.py and the playbacktest.py. The recordtest.py worked but the generated file did not contain the recorded sound. It was just a loud noise sine wave type.

lemorian, 2016-11-29:
There is an error in playback python script.
The PCM function is takes device as string argument and card as number. The code should look like this.
 out = alsaaudio.PCM(alsaaudio.PCM_PLAYBACK,device=card)

closed 2017-02-24 by larsimmisch