larsimmisch/pyalsaaudio #19: failure in test.py

terry1769, 2016-06-23:
cloned from Github today.
Running on a Raspberry Pi 3 with Raspbian Jessie 5/27/2016 build
USB audio card is plugged in.

When I run test.py I get:
...ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

E..ALSA lib confmisc.c:768:(parse_card) cannot find card 'default'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_card_driver returned error: No such device
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_concat returned error: No such device
ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such device
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such device
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default:CARD=default
.
======================================================================
ERROR: testPCM (__main__.PCMTest)
Open a PCM object on every device
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test.py", line 96, in testPCM
    pcm = alsaaudio.PCM(device=pd)
ALSAAudioError: Connection refused [pulse]

----------------------------------------------------------------------
Ran 7 tests in 0.098s

FAILED (errors=1)

lemorian, 2016-11-29:
You probably uninstalled PulseAudio but the entry is still left in alsa config.

BenjaminPelletier, 2016-12-28:
I get the same error, also RPi 3 Raspbian Jessie.  I'm pretty sure I didn't uninstall PulseAudio. sudo apt-get install pulseaudio yields:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
pulseaudio is already the newest version.
pulseaudio set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 209 not upgraded.

Furthermore, running playbacktest.py on a WAV file I already had produced:

ALSA lib conf.c:4705:(snd_config_expand) Unknown parameters CARD=default
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default:CARD=default
Traceback (most recent call last):
  File "playbacktest.py", line 42, in <module>
    out = alsaaudio.PCM(alsaaudio.PCM_PLAYBACK, card=card)
alsaaudio.ALSAAudioError: Invalid argument [default:CARD=default]

larsimmisch, 2017-02-25:
Using the `card` keyword argument is somewhat error-prone. The current version takes a `-d <device>` argument. For the device name, use a name that was output by `aplay -L` or the output of `alsaaudio.pcms()`

closed 2017-02-25 by larsimmisch