larsimmisch/pyalsaaudio #25: ALSAAudioError: No such file or directory [default] running test.py

raspberrypieman, 2016-12-21:
I am having trouble recording in Python using alsaaudio on a Raspberry Pi, so I ran /usr/share/doc/python-alsaaudio/examples/test.py and got:
...E.
===================================
ERROR: testPCM (__main__.PCMTest):
Open a PCM object on every card
------------------------------------------------------
Traceback (most recent call last):
  File "test.py", line 91, in testPCM
    pcm = alsaaudo.PCM(i)
ALSAAudioError: No such file or directory [default] 

------------------------------------------------------
Ran 6 tests in 0.052s
FAILED (errors=1)

I have a Logitech Webcam that I am using to record sound. 
When issue this command: arecord -l it reports:
**** List of CAPTURE Hardware Devices ****
card 1: C920 [HD Pro Webcam C920], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

It records OK, when I use this command:
arecord -D plughw:1 -r 48000 test3.wav

cat /proc/asound/cards reports this:
 0 [ALSA           ]: bcm2835 - bcm2835 ALSA
                      bcm2835 ALSA
 1 [C920           ]: USB-Audio - HD Pro Webcam C920
                      HD Pro Webcam C920 at usb-3f980000.usb-1.3, high speed

larsimmisch, 2017-02-24:
Don't worry about the test output, the test doesn't work well with a quite a few ALSA configurations.

If you can record audio in this configuration using `recordtest.py -d plughw:1`, then you should be fine.

larsimmisch, 2017-02-25:
Reopen the issue if the problem persists

closed 2017-02-25 by larsimmisch