larsimmisch/pyalsaaudio #28: PCM plays two times faster

Lzrnk, 2017-02-16:
It looks like setrate() dosn't work.
playwav.py plays test file two times faster than should do.
Changing rate with setrate() doesn't effect on playing rate.

larsimmisch, 2017-02-22:
Twice as fast sounds a bit like the number of channels, like you have a mono file, but the wav header thinks it's stereo.

In any case, I think the problem is not `setrate`, but more like a strange wav header in your data.

Lzrnk, 2017-02-23:
Thank you for the answer and very useful project.
[test3.wav.zip](https://github.com/larsimmisch/pyalsaaudio/files/795872/test3.wav.zip)

I've checked my file and haven't seen nothing special.
Yes it's mono, but  pcm.getnchannels() retuns 1 as it should be.
Also Audacity and pymedia play  this files without any problem.

larsimmisch, 2017-02-23:
Hmmm. The Python `wave` module also shows it's mono:
```
>>> import wave
>>> w = wave.open('test3.wav')
>>> w.getnchannels()
1
>>> w.getframerate()
48000
```

I'll have to try to reproduce this bug here (don't hold your breath)

larsimmisch, 2017-02-25:
I can not reproduce this. The file does sound fast (like from a TTS with no prosody), but it sounds the same on my Mac, through `aplay` and through `playwav.py`.

Also, the file should play for roughly 6 seconds according to the filesize (576044 / 48000 / 2)

closed 2017-02-25 by larsimmisch