larsimmisch/pyalsaaudio #29: Version 0.8.3 broke Python 3

renekliment, 2017-02-24:
Hey. With the update I'm getting 

```
Python 3.5.3 (default, Feb  2 2017, 16:08:30) 
[GCC 6.3.1 20170109] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import alsaaudio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: /usr/lib/python3.5/site-packages/alsaaudio.cpython-35m-x86_64-linux-gnu.so: undefined symbol: PyInt_FromLong
```

and the same with Python 3.6. The same happens on Travis CI (that's how I've noticed, because we don't have a version locked there), so this is not just my machine.

renekliment, 2017-02-24:
I believe the problem is here: https://github.com/larsimmisch/pyalsaaudio/commit/2314aaeb7ee856317b576e867fa59fd87510bfdb#diff-c173c43425f546cb50e217c77dda54b5R190

And the solution here: http://python3porting.com/cextensions.html#changes-in-python

Also, I suggest at least simple CI tests. Like having Travis CI call `python -c "import alsaaaudio"` for all relevant python versions.

closed 2017-02-24 by larsimmisch

renekliment, 2017-02-24:
Dope, thanks! Can't wait for an updated pypi package so alexa-pi/AlexaPi builds & works again :smile:

larsimmisch, 2017-02-24:
I'll probably make a new release tonight.

renekliment, 2017-02-25:
Aaand it works / builds again, thank you!

commit f19d139f (2017-02-24): Fix C-API usage for Python 3. Closes #29