larsimmisch/pyalsaaudio #31: Undefined behavior for read()

chriskuech, 2017-03-05:
```
import alsaaudio
mic = alsaaudio.PCM(type=alsaaudio.PCM_CAPTURE, device='hw:1,0')
mic.setperiodsize(44100 / 60)
while True:
    (n, data) = mic.read()
```

When run on Raspbian, sometimes `n == -32`, which is undefined behavior according to the documentation

larsimmisch, 2017-03-05:
-32 looks like `BROKEN_PIPE`.

JohnCC330, 2017-03-17:
Just hit the same issue:

This is actually documented in the comments of [alsaaudio.c](https://github.com/larsimmisch/pyalsaaudio/blob/master/alsaaudio.c). However, the [Documentation](https://larsimmisch.github.io/pyalsaaudio/libalsaaudio.html#module-alsaaudio) does not mention this (maybe it should?)

> In case of an overrun, this function will return a negative size: -EPIPE.\n\
> This indicates that data was lost, even if the operation itself succeeded.\n\
> Try using a larger periodsize");
>

larsimmisch, 2017-03-17:
Yes, the documentation should mention that. It's a nasty pitfall.

Every time I stumble across this, I think: we should really throw an exception in this case. But it isn't an unrecoverable error, and throwing an exception would make the data loss worse, so that's not the right thing either.

larsimmisch, 2017-03-17:
Fixed (i.e. documented) in adc0d800e1e9547c351151fc1135a62fc2f66351

closed 2017-03-17 by larsimmisch

JohnCC330, 2017-03-18:
Thanks, Lars!