larsimmisch/pyalsaaudio #32: Non-blocking capture, last frame erratically wrong

JohnCC330, 2017-03-28:
When capturing (in full-duplex), I detected glitches. Further investigation showed those glitches to coincide with the last frame of a captured chunk. This error seems to appear about 80-85% of the time.
[image]
Suspecting incorrect data conversion on my side, I logged the raw chunks on disk, and confirmed that the values were clearly incorrect. 
Also, I never had an 'overrun' (or other) error reported. 

I removed all logging/printing to eliminate timing problems - no difference. The randomness of the problem apparition, but the fact that only one frame is incorrect feels like a timing problem.
[image]
Left column is the sample number, followed by the length of the capture chunk reported by read().
Rest of the line are raw bytes from read, green for those which are Ok, orange are in error. Each frame contains 4 bytes (little endian, stereo).

Any suggestion where to look? I converted the program into a similar version in C, where it seems to work without problem.

JohnCC330, 2017-03-28:
To be clear, the right part is near the end of the chunks (I've frozen the left two columns). The top line indicate the sample number (left/right still interleaved).

larsimmisch, 2017-03-28:
First of all, you win the (nonexistent) price for the best bug report in this project.

Second: Uh huh. I would try to work with a longer periodsize first (I know, this leads to higher latency). 

I think with Python, you will not get inter-thread context switches as smooth as with C, because of the GIL (global interpreter lock), but the details are hazy. So, if your application is multithreaded, this is something to think about.

If I were to implement a duplex playback/recording application on top of pyalsaaudio, I would very likely try to use poll/select and a single thread. There is unfortunately no example yet in the repo.

JohnCC330, 2017-03-28:
> First of all, you win the (nonexistent) price for the best bug report in this project.
;-)

There are no threads at all in the program - this was meant to be first step in a larger project. The machine has 8 cores, so most, if not all other programs are running on other cores. I doubt this is a (lack of) speed problem. 

> Returns a tuple of (file descriptor, eventmask) that can be used to wait for changes on the mixer with select.poll.
Can capture/playback be done with poll/select? I believe I have seen those in the original C ALSA api, but not in pyalsaaudio.

larsimmisch, 2017-03-28:
> Can capture/playback be done with poll/select? I believe I have seen those in the original C ALSA api, but not in pyalsaaudio.

Yes, there is `PCM.polldescriptors` (note to self: document it!)

If you are not using threads and not using poll, how do you do duplex? By interleaving reads and writes?

JohnCC330, 2017-03-29:
> Yes, there is PCM.polldescriptors (note to self: document it!)

Ah... I really should get into the source code it seems - too many hidden features.

> If you are not using threads and not using poll, how do you do duplex? By interleaving reads and writes?

Yes, and I did the same in C, it worked great there.

JohnCC330, 2017-03-29:
Any idea of the file descriptor from the poll descriptor could be used to connect to with an io_add_watch? That would be very elegant. I did a simple test, but the callback is never called:

```
        in_fd = in_pcm.polldescriptors()[0][0]
        in_watch = GObject.io_add_watch(in_fd,
                                        GObject.IO_IN,
                                        self.in_callback)

```

larsimmisch, 2017-03-29:
You are relying on the fact that the `POLLIN` is at position `0`. That is not a sane assumption.

`polldescriptors` returns an array of `(fd, eventmask)`, where eventmask should match up with https://docs.python.org/2/library/select.html#poll-objects.

I think you can do what you want if you match poll event masks to gevent. `POLLIN` is probably `GObject.IO_IN`, maybe the rest lines up neatly as well.

JohnCC330, 2017-03-30:
Ok. Got it working, using the PCM.polldescriptors No more missing samples. Just some more missing hair ;-)

larsimmisch, 2017-03-30:
I'll leave this open until I have documented `polldescriptors` on PCM objects.

closed 2017-03-30 by larsimmisch

JohnCC330, 2017-03-31:
Lars, wouldn't it be useful to:

- Mention that in case of capture channels, it is necessary to start the capturing by doing a read() before the poll/select process?

- Add a method wrapping snd_pcm_start() into alsaaudio, to do the same (but more elegantly)?

commit eca217df (2017-03-30): Document `PCM.polldescriptors`.

Closes #32