larsimmisch/pyalsaaudio #33: Missing sample(s) at the end of 'chunks'

JohnCC330, 2017-04-11:
The problem reported in a previous issue remains. I've done quite a few tests, capturing samples and reading them in several different ways: 

- Tight loops
- Threads
- Poll/Epoll

Each time, samples are lost at chunk boundaries. I've changed sample frequencies from 12000 to 48000 Hz. Change chunksize to different values. Nothing seems to solve the problem. I did a C version of the 'tight loop', which seems to work flawlessly. I can't really imagine that Python wouldn't be able to handle 12000 samples/s, so I don't know if there is some logic mine which is failing.

I'd appreciate pointers to continue experimenting!

Notes on the following program:

- The input/outputs are set at the top. In my case they correspond to Front Mic and Speaker output. I've connected them together with a cable to test. It's not necessary to do that, but probably it'd good to connect something to the input to see the missing samples.
- Two methods are in the Generator class. One for Poll, the other uses threads. Select the desired one near the bottom by (un)commenting (in `testcalls()`).
- The program generates a 1 second 1000 Hz tone and plots the result using matplotlib.

```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
#
#  lms_sines.py
#
#  Copyright 2017 John Coppens <john@jcoppens.com>
#
#  This program is free software; you can redistribute it and/or modify
#  it under the terms of the GNU General Public License as published by
#  the Free Software Foundation; either version 2 of the License, or
#  (at your option) any later version.
#
#  This program is distributed in the hope that it will be useful,
#  but WITHOUT ANY WARRANTY; without even the implied warranty of
#  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
#  GNU General Public License for more details.
#
#  You should have received a copy of the GNU General Public License
#  along with this program; if not, write to the Free Software
#  Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston,
#  MA 02110-1301, USA.
#
#

import alsaaudio
import pylab as plt
import math as m
import numpy as np
import scipy as sp
import cmath
import select
import struct
from   threading import Thread
import sys, time
import pdb

debug = False

MIC_INPUT   = 2
SPKR_OUTPUT = 8
LOG_FNAME   = "test.dat" # None

class In_Thread(Thread):
    def __init__(self, in_pcm, in_frames, channels):
        super(In_Thread, self).__init__()
        self.in_pcm = in_pcm
        self.in_data = []
        self.in_frames = in_frames
        self.channels = channels

    def run(self):
        in_frame = 0
        while in_frame < self.in_frames:
[... 163 lines cut ...]

        plt.xlabel("Time (%s)" % ("s" if plot_time else "samples"))
        plt.ylabel("Signal (16bit signed)")
        plt.grid(True)
        plt.show()

    def print_data(self, data):
        for y in data:
            print("%6d " % y, end = '')
        print()

def test_calls():
    gen = Generator(chunksize = 64, fs = 48000)
    #data = gen.selfcapture_threads(1000, 48000, channels = 2)
    data = gen.selfcapture_poll(1000, 48000, channels = 2)
    gen.plot_data(data, channels = 2, plot_time = False)

def main(args):
    test_calls()
    return 0

if __name__ == '__main__':
    import sys
    sys.exit(main(sys.argv))
```

JohnCC330, 2017-04-12:
In the code above, lots of code is duplicated because of the two methods (threads and polling) implemented.

Here are some screenshots of the output. These is the generated sine wave, reproduced and (using a cable) re-introduced (to the left and right mic inputs), recorded and plotted:

[image]
(This is a single lost sample, the value from 'read' was 0 for both channels)

[image]
(An example of a double error)

To be sure the sine was was generated correctly, I connected the output to another machine, and recorded it there using Audacity.

ossilator, 2022-08-05:
is this still reproducible?

RonaldAJ, 2022-08-26:
It claims to be python3 code, so we can expect it to work. Have you tried? 

There is a small chance that the fix I provided for #51 also fixes this. But I would have to spend more time to see whether these problems are the same.

RonaldAJ, 2022-08-26:
With the changes below the OP code runs again. But the figures don't reproduce on my system, this might require some alsa configuration to work.

`#!/usr/bin/env python3
# -*- coding: utf-8 -*-
#
#  lms_sines.py
#
#  Copyright 2017 John Coppens <john@jcoppens.com>
#
#  This program is free software; you can redistribute it and/or modify
#  it under the terms of the GNU General Public License as published by
#  the Free Software Foundation; either version 2 of the License, or
#  (at your option) any later version.
#
#  This program is distributed in the hope that it will be useful,
#  but WITHOUT ANY WARRANTY; without even the implied warranty of
#  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
#  GNU General Public License for more details.
#
#  You should have received a copy of the GNU General Public License
#  along with this program; if not, write to the Free Software
#  Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston,
#  MA 02110-1301, USA.
#
#

import alsaaudio
import pylab as plt
import math as m
import numpy as np
import scipy as sp
import cmath
import select
import struct
from   threading import Thread
import sys, time
import pdb

debug = True

MIC_INPUT   = 2
SPKR_OUTPUT = 8
LOG_FNAME   = "test.dat" # None

class In_Thread(Thread):
    def __init__(self, in_pcm, in_frames, channels):
        super(In_Thread, self).__init__()
        self.in_pcm = in_pcm
        self.in_data = []
        self.in_frames = in_frames
        self.channels = channels

    def run(self):
        in_frame = 0
        while in_frame < self.in_frames:
            l, chunk = self.in_pcm.read()
            if l > 0:
                d = list(struct.unpack("<" + "h" * (l * self.channels), chunk))
                self.in_data += d
                in_frame += l
        return

class Out_Thread(Thread):
    def __init__(self, out_pcm, out_data, out_frames, chunksize):
        super(Out_Thread, self).__init__()
        self.out_data = out_data
        self.out_frames = out_frames
        self.out_pcm = out_pcm
        self.chunksize = chunksize

[... 157 lines cut ...]

        plt.xlabel("Time (%s)" % ("s" if plot_time else "samples"))
        plt.ylabel("Signal (16bit signed)")
        plt.grid(True)
        plt.show()

    def print_data(self, data):
        for y in data:
            print("%6d " % y, end = '')
        print()

def test_calls():
    gen = Generator(chunksize = 64, fs = 48000)
    #data = gen.selfcapture_threads(1000, 48000, channels = 2)
    data = gen.selfcapture_poll(1000, 48000, channels = 2)
    gen.plot_data(data, channels = 2, plot_time = False)

def main(args):
    test_calls()
    return 0

if __name__ == '__main__':
    import sys
    sys.exit(main(sys.argv))
`

RonaldAJ, 2022-08-26:
The previous issue mentioned is probably: #32

ossilator, 2024-02-01:
as OP apparently lost interest, i'm closing this.

i didn't study the test code in detail, because it's _way_ too complex. - if there is an actual problem in the library, it can be reproduced _much_ more easily. if the complexity seems necessary for reproduction, then either the bug is in the user code, or the real trigger is most likely system load of some kind (cpu, memory, device activity (interrupts), etc.).

the pattern in the plots does not indicate that samples are lost or added, but rather that the buffer is overwritten or not written in the first place. i suspect a bug in a lower layer (hardware, driver, libalsa), and hope it was fixed or worked around in the right place meanwhile.

closed 2024-02-01 by ossilator