larsimmisch/pyalsaaudio #35: Allowable period size seems too restrictive

bewantbe, 2017-05-08:
For recording, when set period size (by `.setperiodsize`) to a reasonable value which is used in arecord (get from `$ arecord -v --dump-hw-params -D 'default' -f S24_3LE -r 48000 -c 2 --duration=1 /dev/null`), e.g. periodsize = 6000. I get an Exception:

    ALSAAudioError: Capture data too large. Try decreasing period size

This is confusing and annoying, because I'm providing valid parameters...

I found (seems) related code here (https://github.com/larsimmisch/pyalsaaudio/blob/master/alsaaudio.c#L772):

    char buffer[8000];

    if (self->framesize * self->periodsize > 8000) {

        PyErr_SetString(ALSAAudioError,"Capture data too large. "
                        "Try decreasing period size");

It would be great if the limit of period size set to a device depend value (instead of hard coded).

larsimmisch, 2017-05-08:
I agree. Maybe I'll get to it this week.

Note to self: allocate `PyBytes`/`PyString` directly?

jcea, 2018-02-14:
I am hitting this problem too.

I am currently modifying the sourcecode to increase the buffer size.

Since the buffer is dynamically allocated in the stack, this is cheap. The allocation, that is. Then we have two copies: from kernel to buffer, from buffer y to python byte object. The amortized cost is constant, because if your period size is double, you call "read()" half the times.

jcea, 2018-02-15:
I don't see the need to impose a hard limit in the allowable period size beside what ALSA limit be.

I see two approaches with this:

1. Use `alloca()` to request memory in the stack. This is very cheap and you avoid the statically defined buffer size. Performance would be quite equivalent to current code, but no statically max size for the buffer. See, for instance: https://binarydebt.wordpress.com/2013/09/21/dynamic-memory-allocation-on-the-stack/

2. Preallocate a Python bytes object with `PyBytes_FromStringAndSize(NULL, len)`. Then use `PyBytes_AsString()` to get access to the internal memory of the object and use it as a target for `pcm_readi()`. This has the advantage of avoiding a copy operation (from "ALSA -> Buffer -> Python object"  to "ALSA -> Python object"). If the read fails, we can use `_PyBytes_Resize()` to convert the string to a empty string, or we could dispose that object and return a preallocated "empty" bytes object. We do more work if we have an overrun, but if you have frequent overruns you are in big trouble anyway. Details: https://docs.python.org/3.6/c-api/bytes.html#c.PyBytes_FromStringAndSize. Python 2.7 has similar API: https://docs.python.org/2.7/c-api/string.html

Do you want a Pull request for this? :)

larsimmisch, 2018-02-15:
Yup. Do send a PR. I think I meant to do something like 2) once, but in the end didn't do it because of one niggly detil that I couldn't be bothered with.

bewantbe, 2018-02-16:
Just to add a use case that a larger periodsize can help (which was omitted in my original post):

I was writing a program that supports FFT (For computing audio spectrum) length up to 65536, which is a common size. This corresponds to a period size = 65536 * 2Byte * 2Channel = 256KB. Note that the ALSA supports periodsize up to 1.3MB for my sound card.

Of course, I can concatenate these short 8KB data to a consecutive 256KB data, then perform the FFT. I did that at last, but that is inconvenient:

1. It needs some more code to deal with the concatenation, this also cost more CPU time and wakes up CPU more often (not energy efficient). Wow... I'm talking about energy efficient and python?

2. The 1.37 sec (65536/48KHz) sample data might need more that 41ms (8KB/2/2/48KHz) to process, i.e. the 8KBx2(or x3, x4?) buffer might go overrun. To prevent overrun, the recording and processing code need to run on different threads. This adds more code complexity.

To summary, larger periodsize can keep the simple program simple.

jcea, 2018-02-20:
@bewantbe, could you possibly try PR  #53 ?

larsimmisch, 2018-02-20:
@bewantbe I've merged, #53, so can you try master instead?

bewantbe, 2018-02-27:
Tested rev: 723eff38, for python2 version, work as expected. Great!

Just, a very minor problem: there is not limit to `.setperiodsize`, so `.read()` could always return fewer samples (due to hard limit in ALSA or hardware). And there seems no warning or corresponding `.getperiodsize()` to verify if the `.setperiodsize()` fully succeeded.

larsimmisch, 2018-02-27:
Thanks for testing the latest version, that is good news. I intend to fix a few other bugs before I make a new release, but it is planned soon(ish).

I just pushed https://github.com/larsimmisch/pyalsaaudio/commit/1aae655d24339d3fd07e7a98c64b867365202dff to address your concerns.

The idea is:  if `setperiodsize` fails, I expect the call to `alsapcm_setup` to fail. 

I don't think this is what you had in mind, but if ALSA does not honour its API, there is little we can do about it.

bewantbe, 2018-03-05:
I did some more test (see attached file), the problem is `snd_pcm_hw_params` won't return error even if `snd_pcm_hw_params_set_periods` failed. The variable `snd_pcm_hw_params_t*` will contain a valid but different parameter set, and `snd_pcm_hw_params` will success.
e.g.

set:
- period_size: 262144
- periods: 4

get:
- snd_pcm_hw_params_set_periods error.
- period_size = 262144
- periods = 2
- `snd_pcm_start` and `snd_pcm_readi` run without a problem.

set:
- period_size: 262145
- periods: 4

get:
- snd_pcm_hw_params_set_period_size error.
- period_size = 131072
- periods = 4
- `snd_pcm_start` and `snd_pcm_readi` run without a problem.

Seems that in order to detect error, we must check every return value of these functions....
[rec_test.c.tar.gz](https://github.com/larsimmisch/pyalsaaudio/files/1781722/rec_test.c.tar.gz)

bewantbe, 2018-03-07:
OK, I saw that the function `setperiodsize` will return actual periodsize. I can do something like:
```python
ps_request = 1000000
ps_ret = pcm.setperiodsize(ps_request)
if ps_ret < ps_request:
    print("periodsize too large")
```
So, I think this problem can be viewed as solved, similar for other `set*` functions. Maybe just need to mention it in the document.

omarmarcillo, 2018-08-11:
was there a final solution for setting a larger period size?,

I am having an overrun issue that I think is related to the period size. I am using a octosound card (6 input channels) with a raspberry pi 3B and I am sampling at 32 ksps. I cannot set a period size higher than 600 (I think. this is related to a 4092 maximum buffer size). If I use 600 for period size after reading the first block of data I immediately get an overrun value from read. Then the CPU consumption of my Pi goes to almost 340%, I think this is related to the overrun status. Setting a higher period may solve my problem.

bewantbe, 2018-08-11:
@omarmarcillo The period size now can be set to a larger value, see my comment on [Feb 28](https://github.com/larsimmisch/pyalsaaudio/issues/35#issuecomment-369064484) and [Mar 7](https://github.com/larsimmisch/pyalsaaudio/issues/35#issuecomment-371144793).

But you need to use the one in this [repo](https://github.com/larsimmisch/pyalsaaudio), **NOT** the one from pip or repository of your Linux distribution. See [Manual installation](https://github.com/larsimmisch/pyalsaaudio#manual-installation).

larsimmisch, 2018-08-11:
Thanks. Yes, you need to install from source from this repo. I'll do a new release hopefully soon, but there are some other outstanding issues I want to fix first.

closed 2018-08-11 by larsimmisch

larsimmisch, 2018-08-13:
Yes. I would expect that creating a thread for each read would work poorly.

You could create a single thread for reading and let it block on read. Or you could use event multiplexing and use ALSA in nonblocking mode (another term to google for is _reactor pattern_)

omarmarcillo, 2018-08-13:
I would start by removing the macros for the starting the thread (Py_BEGIN_ALLOW_THREADS) and running in the main program. AS I am calling the function every ~10-20 ms (period-size 1600 and sampling rate 44100Hz) I would think the overhead of the function would be minimal relative to the time of data to be ready.

larsimmisch, 2018-08-13:
No.

`Py_BEGIN_ALLOW_THREADS` *does not* start a thread. It is a C-level Macro that manages the GIL. There is no reason to modify the C code for your requirements unless you wish to shoot yourself in the foot.

You need to think about threading on the *Python* level. As I said, there are a few ways to do it: single python thread for reading, event multiplexing, etc.