larsimmisch/pyalsaaudio #36: Help for implement `aplay -l` - List audio interfaces

SrMouraSilva, 2017-05-09:
Hello, I would like a help for me to implement the similar feature to `aplay -l`.
Could you guide me?

Contextualization: I need in my application to list the audio interfaces. Do something like `qjackctrl`. I found a solution to manually pick up from `aplay -l`, but this raises problems in linux with other defined languages.

[image]

larsimmisch, 2017-05-09:
Please have a look at https://larsimmisch.github.io/pyalsaaudio/libalsaaudio.html

`alsaaudio.pcms(alsaaudio. PCM_CAPTURE)` is roughly equivalent to `aplay -L`, not sure how to get the result of `aplay -l`

closed 2017-05-09 by larsimmisch

SrMouraSilva, 2017-05-10:
Hi, thanks for your feedback.

I had already checked the use of `alsaaudio.pcms`. However, it (which corresponds to `aplay -L`) does not enumerate the audio interfaces. I need something similar to `aplay -l`.

I intended to support your project by adding this functionality. But since I have no idea how to get this information and you do not, unfortunately I will not be able to contribute.