larsimmisch/pyalsaaudio #43: Problem with playback on single channel

interlocuteur, 2017-12-05:
I'm using pyAlsaAudio for a tone generator application.
Versions are 0.8.4 for pyAlsaAudio (installed through pip) and python is 3.5.
I set the following parameters:
- rate: 8KHz
- format: U8
- periodsize: 100
- channels: 1
It works but:
- there is some noise ('pop') at the beginning of the playback
- the latency is very long

I was surprised with the latency since Alsa and pyAlsaAudio are designed for low-latency.
I used dumpinfo() function to diagnose and found that setchannel() to 1 generate weird settings:
- period time = 131072000 us
- period size = 1048576 frames
- buffer time = 0 us
- buffer size = 4194304 frames
The buffer size probably explain the long latency.

It looks like an overflow problem or some un-initialized variable.
I tried to dig into alsaaudio.c (found on github) and can't find the problem.

Note: the latency problem and the 'pop' noise do not happen when channels=2

By the way and unrelated, dumpinfo() reports 'period time' twice. I suppose the intent of the second one was to print some other variable.

ossilator, 2022-08-05:
is this reproducible with v0.9, using the new method to paremeterize the pcm object?

ossilator, 2024-02-01:
this was presumably fixed along with #63.

closed 2024-02-01 by ossilator