larsimmisch/pyalsaaudio #45: `alsaaudio.ALSAAudioError: Invalid argument` with BlueALSA due to no matching configuration

follower, 2018-01-05:
While this comment says "don't care if any of this fails" it *is* possible for a failure to prevent the device being opened:

<https://github.com/larsimmisch/pyalsaaudio/blob/bfe48997215d87ac939e408192982a88dd5843c5/alsaaudio.c#L319-L320>

The issue seems to be (a) in part related to `res` never being overwritten if a failure occurs; and, (b) some devices not handling requests for "exact" values:

<https://github.com/larsimmisch/pyalsaaudio/blob/bfe48997215d87ac939e408192982a88dd5843c5/alsaaudio.c#L336>

The resulting error in my case looked like:

```
    out = alsaaudio.PCM(alsaaudio.PCM_PLAYBACK, device=device)

...

alsaaudio.ALSAAudioError: Invalid argument [bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp]
```

In my case I worked around the issue by changing the following lines in `alsapcm_new()`:

```
    self->rate = 48000;
    // ...
    self->periodsize = 4000;
```

I think the "correct" approach is to use the approach used by [`pyalsa.pyx`](https://pypkg.com/pypi/pyalsa/f/) which uses the `_near` suffixed functions:

```
snd_pcm_hw_params_set_rate_near(self.pcm, params, &rrate, NULL)
```

(Once I got `pyalsa` (from BigAsterisk.com) to compile (which required changing `__new__` to `__cinit__`) I had no issues using it to connect to the Bluetooth speaker device--it also seems to have more provision for informative error messages. :) )

[Ideally I'd provide more detail/patch but this has taken many hours to track down, so thought I'd create a quick issue now in case anyone else is affected in future rather than "intend" to do something more thorough but not quite get there. :) ]

follower, 2018-01-05:
Machine configuration:

  * Raspberry Pi 3
  * Raspbian Stretch Lite (`2017-11-29-raspbian-stretch-lite.img`)

Device configuration:

 * "CANZ 808" Bluetooth speaker

larsimmisch, 2020-04-23:
The fix in https://github.com/larsimmisch/pyalsaaudio/commit/8ed9f924cdd990704f206572611d04859095f36d is not tested yet

larsimmisch, 2020-07-16:
Successfully tested with Bluetooth speakers, hope that was it.

commit 8ed9f924 (2020-04-23): Attempt to fix #45
closed 2020-07-16 by larsimmisch