larsimmisch/pyalsaaudio #48: getvolume() Question

commoner1945, 2018-02-11:
This is more of question than  issue. Apologize posting here because I can't find place to discuss my question. 

Testing with pyalsaaudio v0.8.4 with code below
```
import alsaaudio

mixer = alsaaudio.Mixer()
vol = mixer.getvolume()
print(vol)
```
It prints  [27, 27]. But checking with alsamixer, it shows Master volume value is 20. How should I interpret values obtained by getvolume() method? 

Thanks

cuu, 2018-05-02:
I have the same issue...

matthijskooijman, 2020-06-16:
I believe that pyalsaaudio returns linear sound intensity (e.g. if you have two identical speakers and disconnect one, the intensity halves, same if you make a sine wave peak amplitude in your wav file half as big), while alsamixer shows a logarithmic value (which more closely resembles the human ear, we need a lot more than double intensity to perceive something as twice as loud).

Often the algorithmic value is called SPL and measured in dB, but I suspect that alsamixer does some extra tricks to scale the db value back into what seems to be a percentage (maybe it scales from 0dB to -100dB?).

matthijskooijman, 2020-06-16:
Also see issue #8 which has a lot more detail about this conversion and scaling.

quantenschaum, 2020-12-18:
There are several possible units in alsa to express volume in. 

>%, dB or units that are specific for the card/device

The `amixer` cli tool allows to use all of them, but pyalsaaudio doesn't. 

It would be nice, if pyalsaaudio allowed the use of all of these units or at least always use  the device specific units. The device specific range is exposed via `Mixer.getrange()` but `Mixer.setvolume()` does not accept these units. Solving this would also solve #8.

larsimmisch, 2022-03-28:
Resolved by #109

closed 2022-03-28 by larsimmisch