larsimmisch/pyalsaaudio #49: "PCM.dumpinfo()" is undocumented

jcea, 2018-02-14:
Trying to debug an issue I bumped on "PCM.dumpinfo()". There is no documentation about it in https://larsimmisch.github.io/pyalsaaudio/libalsaaudio.html

I get this information:

```
>>> a.dumpinfo()
PCM handle name = 'hw:CARD=SB,DEV=0'                                           
PCM state = PREPARED
access type = RW_INTERLEAVED
format = 'S16_LE' (Signed 16 bit Little Endian)
subformat = 'STD' (Standard)
channels = 2
rate = 44100 bps
period time = 43537 us
period size = 1920 frames
buffer time = 0 us
buffer size = 7680 frames
periods per buffer = 0 frames
exact rate = 44100/1 bps
significant bits = 16
period time = 43537 us
is batch = 0
is block transfer = 1
is double = 0
is half duplex = 0
is joint duplex = 0
can overrange = 0
can mmap = 1
can pause = 1
can resume = 0
can sync start = 1
```

jcea, 2018-02-15:
Would be interesting if those details were available also as a dictionary or named tuple.

larsimmisch, 2020-04-05:
No, it isn't, but see https://github.com/larsimmisch/pyalsaaudio/pull/82, which gives at least some information. That was recently merged to master (not released yet).

closed 2020-04-05 by larsimmisch