larsimmisch/pyalsaaudio #50: "PCM.read()" returns wrong data when overrun

jcea, 2018-02-14:
When `PCM.read()[0]` is -32, we had an overrun. Nevertheless the method returns a "random" length string:

```
>>> w=a.read(); print(w[0], len(w[1]))   # We start capture,
1920 7680
>>> w=a.read(); print(w[0], len(w[1]))   # Overrun. The stream state changes to "PREPARED".
-32 113
>>> w=a.read(); print(w[0], len(w[1]))   # We start capture,
1920 7680
>>> w=a.read(); print(w[0], len(w[1]))   # Overrun. The stream state changes to "PREPARED".
-32 85
>>> w=a.read(); print(w[0], len(w[1]))
1920 7680
>>> w=a.read(); print(w[0], len(w[1]))
-32 1
>>> w=a.read(); print(w[0], len(w[1]))
1920 7680
>>> w=a.read(); print(w[0], len(w[1]))
-32 46
```

Studying the sourcecode, I see this:

```
    if (res == -EPIPE)
    {
        /* EPIPE means overrun */
        snd_pcm_prepare(self->handle);
    }
    Py_END_ALLOW_THREADS
....
  return Py_BuildValue("iy#", res, buffer, res*self->framesize);
```

So, when res=-32 (EPIPE), we copy `res*self->framesize` bytes from `buffer`. This value is negative. Apparently Python interprets this as "copy until the first ZERO, not include it".

Expected outcome: If an overrun happened, the returned string should be empty or the object None. Not whatever happened to be in the buffer: random memory (if you used the stack a lot) or the previous (probably corrupted and truncated to the first ZERO) frame period data.

Do you want a pull request?.

larsimmisch, 2018-02-14:
Yes, please send a PR.

jcea, 2018-02-19:
Done!. PR #52

closed 2018-02-20 by larsimmisch

jiangcheng88, 2019-11-13:
I think we can read faster or new a thread to read data from alsaaudio .  We can avoid this problem .