larsimmisch/pyalsaaudio #51: Buffering is hardcoded to 4*periodsize

jcea, 2018-02-14:
I do 48000 samples per second, 2 channels, 16 bits per channels. Since current pyalsaaudio limits the `PCM.read()` buffer to 8000 bytes (see issue #35 ), I am using a period size of 1920 frames. That is, each `PCM.read()` returns 1920 frames, or 7680 bytes (less that the limit of 8000 bytes). I must do a `PCM.read()` every 40 ms to avoid overuns. it should be not a problem.

I am doing this in a Raspberry PI 3 and since I upgraded from Debian Jessie to Debian Stretch I am having several overruns per day (this thing is recording and playing 24/7). The code is exactly the same, even the python interpreter is the same (compiled from source by myself). Pyalsaaudio version is the same also. I don't know the reason, suggestions are welcomed.

Studying the sourcecode to improve the situation, I see this:

```
  snd_pcm_hw_params_set_period_size(self->handle, hwparams,
                                      self->periodsize, dir);
  snd_pcm_hw_params_set_periods(self->handle, hwparams, 4, 0);
```

Apparently pyalsaaudio is instructing ALSA to use a buffer of 4*`period_size`. In my case, 160 ms. This value is hardcoded and not documented anywhere. I think this should be documented somewhere and, hopefully, configurable.

larsimmisch, 2018-02-14:
That looks fishy. I'll take a look.

eraoul, 2018-04-19:
Any updates on this one? I noticed a similar thing:

I am trying to record audio, using format PCM_FORMAT_U24_LE (which I expect to be 3 bytes per channel-sample), a periodsize of 256, and 4 channels.

When I call "read" I'm getting back a buffer of 4096 bytes. I expected 3 bytes for the U24 * 4 channels * 256 periodsize = 3072. 

If I change the format to PCM_FORMAT_FLOAT_LE to get 4-byte floats out instead, I also get a buffer of 4096 bytes back from read(). So the format isn't affecting the buffer size, and indeed it looks like it's hardcoded to something like period_size * 4 * #channels. 

Any idea what's going on here?

mcherep, 2019-10-23:
It's been a long time since the issue was reported, and I seem to find the same behaviour. Any plans to fix this? Thanks!

RonaldAJ, 2020-09-30:
The behaviour also affects the recordtest.py example. There the direct dump of the whole buffer into to the output file leads to inclusion of invalid data. And as an example it raises the expectation that you you only get valid examples, which I think would be the most logical behavior.

RonaldAJ, 2020-09-30:
I think the problem can large be solved by resizing the buffer in alsapcm_read. But I have no experience coding Python extension directly in c.  

But this snippet:
`if (res <= 0) {
#if PY_MAJOR_VERSION < 3
		/* If the following fails, it will free the object */
		if (_PyString_Resize(&buffer_obj, 0))
			return NULL;
#else
		/* If the following fails, it will free the object */
		if (_PyBytes_Resize(&buffer_obj, 0))
			return NULL;
#endif
	}`

can probably be replaced with something along the following lines:

`        int newsize;

        if (res <= 0) {
            newsize = 0;
        } else {
            newsize = res * self->framesize;
        }

#if PY_MAJOR_VERSION < 3
		/* If the following fails, it will free the object */
		if (_PyString_Resize(&buffer_obj, newsize))
			return NULL;
#else
		/* If the following fails, it will free the object */
		if (_PyBytes_Resize(&buffer_obj, newsize))
			return NULL;
#endif`

But the documentation on _PyBytes_Resize is somewhat obscure to me.

commit 1e3c7f3f (2020-10-27): Fix #51: Only return valid part of the buffer in the read function
commit 5c481b40 (2020-10-27): Fix #51: Only return valid part of the buffer in the read function; avoid unnecesssary work by only changing size when needed
commit d0372976 (2020-10-27): Merge pull request #91 from soundappraisal/master

Fix #51: Only return valid part of the buffer in the read function
closed 2020-10-27 by larsimmisch