larsimmisch/pyalsaaudio #56: Init Problems with automatic Start after Reboot

BorisJB, 2018-02-26:
Hi! I am using this module for my Web-Radio Project on a Raspi Zero W.
All works fine, when I start my python program from the console.
But I have problems if I start the script automatically after reboot.

**Test Script:**
import sys
import alsaaudio

kwargs = {}
mixers = alsaaudio.mixers(**kwargs)
print( "mixers ", mixers)

Started manually gives (this also works with 'sudo'):
('mixers ', [u'Master', u'Capture'])

If I start it with /etc/rc.local:
/usr/bin/python /home/pi/alsaTest.py > /home/pi/alsaLog.txt

I can not get a list of mixers:
('mixers ', [])

Maybe somebody has an idea how to make this work automatically after reboot.

Thanks,

Boris

larsimmisch, 2018-02-26:
I guess you have to wait until the service `alsa-utils` (or similar) has been run. It looks like you're not using `systemd`, so maybe just run the necessary `alsactl restore` before you python script?

larsimmisch, 2018-02-26:
Closing this, because it does not seem to be an issue with pAlsaaudio itself.

closed 2018-02-26 by larsimmisch