larsimmisch/pyalsaaudio #57: Support for setting enum values

rnixx, 2018-03-17:
Unfortunately pyalsaaudio does not support setting enum values. Is is planned to add this feature? Does this imply a lot of work or would it be a low hanging fruit?

Thanks

larsimmisch, 2018-03-18:
There is `setenum`, but it isn't a friendly API.

I have no immediate plans to create a good, pythonic API. The last time I seriously thought about it, I decided it was definitely not low-hanging fruit. I don't think it would be hard, though.

PRs are welcome.

larsimmisch, 2018-03-18:
I now remember that I thought a good `setenum`-API would be best written in a python module (as opposed to the C-only module we have now) and I couldn't get myself to introduce this extra complexity just for that feature and I was definitely not in the mood to do it in C.

rnixx, 2018-03-19:
Hi,
looks like setting enum values already has been added by @chrisdiamand here https://github.com/larsimmisch/pyalsaaudio/commit/bf24ec65cac6831769966b4cf0a5680fbb29def6 but it's not reflected in the online docs

rnixx, 2018-03-20:
Just tested ``setenum`` on mixer objects and it works. Please update documentation. Sorry for the noise.

closed 2018-03-20 by rnixx