larsimmisch/pyalsaaudio #59: Segmentation fault

dgomes, 2018-04-30:
OS: Raspbian (Debian Stretch) - ALL packages updated
Kernel: Linux 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux

Happens in both python2 and python3 with pyalsaaudio installed through PIP

Python 3.5.3 (default, Jan 19 2017, 14:11:04) 
[GCC 6.3.0 20170124] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> s[::-3]
KeyboardInterrupt
>>> import alsaaudio
>>> alsaaudio.mixers(device="bluealsa")
Segmentation fault (core dumped)

GDB information:

Core was generated by `python3.5'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  alsamixer_list (self=<optimized out>, args=<optimized out>, kwds=<optimized out>) at alsaaudio.c:1168
1168	alsaaudio.c: No such file or directory.

dgomes, 2018-05-01:
I was able to overcome this issue by disabling profiles in **bluealsa** daemon:

`bluealsa --disable-hsp --disable-hfp`

dgomes, 2018-10-21:
This issue still exists.

For future reference, bluealsa has changed interface. You now need to start with:

`bluealsa -p a2dp-source -p a2dp-sink`

larsimmisch, 2020-07-16:
I can't reproduce this, closing

closed 2020-07-16 by larsimmisch