larsimmisch/pyalsaaudio #60: PCM playback sped up

Frodriguez26, 2018-05-16:
`
This is a great project! Though I have an issue with the play back speed. I have been searching for an answer for a couple weeks now and I haven't found anything yet.

I am using a slightly modified version of the recordtest.py example from the main github page: 

    import alsaaudio
    import time
    
    f = open("out_test.raw", "wb")
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NONBLOCK, cardindex=1

    # Set attributes: Mono, 44100 Hz, 16 bit little endian samples
    inp.setchannels(1)
    inp.setrate(44100)
    inp.setformat(alsaaudio.PCM_FORMAT_S16_LE)
    inp.setperiodsize(160)

    loops = 1000000
    while loops > 0:
        loops -= 1
        # Read data from device
        l, data = inp.read()

        if l:
            f.write(data)
            time.sleep(.001)

and I am also using a modified version of the playbacktest.py examples which looks like:
    
    import time
    import alsaaudio

    index = 1
    f = open('out_test.raw', 'rb')
    # Open the device in playback mode.
    out = alsaaudio.PCM(alsaaudio.PCM_PLAYBACK, cardindex = index)

    # Set attributes: Mono, 44100 Hz, 16 bit little endian frames
    out.setchannels(1)
    out.setrate(44100)
    out.setformat(alsaaudio.PCM_FORMAT_S16_LE)

    # The period size controls the internal number of frames per period.
    # The significance of this parameter is documented in the ALSA api.
    out.setperiodsize(160)

    # Read data from stdin
    data = f.read(320)
    while data:
        out.write(data)
        data = f.read(320)

When I run the recordtest.py, it seems to work fine and I am able to hear the recording with: 

    aplay -D plughw:1 -f S16_LE -r 44100 out_test.raw

But when I try using playbacktest.py file, it plays the sound very fast. I have tried changing the rate, periodsize and buffer size and none of those change the speed at which the file is being played. I have also tried saving the recording into a wave file then playing it using the playwav.py example and it is still sped up.  

I am using a raspberry pi 3 with a usb sound card hooked up to the usb port. Thanks for the help!
`

riverar, 2018-08-28:
Can confirm this on 0.8.4 as well with similar hardware.

aidanhb, 2018-10-25:
Third-ed on 0.8.4 for a PreSonus Audiobox soundcard. Sounds fine on system speakers.

ozett, 2018-12-04:
i have raspberry pi zero W,
sounds play ok with aplay, very fast with default test...
i confirm search fo an answer here..

ShellAddicted, 2019-05-17:
I'll just leave this here, may work or not, (it did on my machine)
Specify the device as `plughw` example: `plughw:1,0`

akamensky, 2021-01-07:
Same on RPi 4B w/ Ubuntu 20.10 64bit.

@ShellAddicted using `plughw` sink makes no difference on my end.

ossilator, 2022-08-05:
is this reproducible with v0.9, using the new method to parameterize the pcm object?

JetA2, 2024-01-03:
Using latest git clone, the playback speed varies with the period size. 
Large period size gives slower playback speed, small period size gives faster playback speed.

Something seems to be broken somewhere...

ossilator, 2024-02-01:
hmm, the plughw hint should have worked if applied correctly, because the most likely reason for the failure is that alsa chose different stream parameters than requested, due to hardware limitations. that shouldn't happen with plughw. this hypothesis can be verified by calling out.dumpinfo().

alternatively, this might have been fixed along with #63, hence my question above.

so i'm closing this as "need more info".

closed 2024-02-01 by ossilator