larsimmisch/pyalsaaudio #64: play sine on either left or right

GOBish, 2018-09-16:
Hi,
I'm trying to generate a sine wave and play it on either the left channel or right channel.
I have been experimenting with the isine code, thank you very much for providing that.
Could you possibly give an example of how I could achieve this?
Also, to implement the isine code I have been using "from isine import change" but then I need a sleep function to get it to play for any time. Is this how it is intended to be used or am I doing it wrong?
Thanks!

kfrncs, 2018-09-18:
These might be of interest to you:

https://github.com/ttm/mass
https://github.com/ttm/music

larsimmisch, 2020-07-13:
To play data on just one channel, you could open a stereo device and set the data for one channel to 0 - typically the data for stereo is interleaved (LRLR off the top of my head). The size of your data and its interpretation depend on the format, though.

The default format is `PCM_FORMAT_S16_LE`. So you could use `struct.pack('hh', 0, v)` to construct a single frame.

ossilator, 2024-02-01:
this question is moderately related to the sine example failing to work as advertised, see #42.

other than that, i think we can close this as "resolved support question".

closed 2024-02-01 by ossilator