larsimmisch/pyalsaaudio #65: Recording only in 96000khz sampling rate

venkat-kittu, 2018-09-17:
I am trying to record audio by using code from recordtest.py but when i am executing the program my audio is recorded in sampling rate of 96000khz only.
I tried keeping samplingrate to 48000, but it is not reflecting in output.
How much sampling rate i kept in program always it is giving output of 96000khz only..

my code is below

## aplay -r 8000 -f S16_LE -c 1 out.raw

#!/usr/bin/env python

from __future__ import print_function

import sys
import time
import getopt
import alsaaudio

if __name__ == '__main__':

    f = open(sys.argv[1], 'wb')
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NONBLOCK, device='hw:0,0')
    inp.setrate(48000)
    inp.setchannels(1)
    inp.setformat(alsaaudio.PCM_FORMAT_S16_LE)
    inp.setperiodsize(160)
    
    loops = 5000
    while loops > 0:
        loops -= 1
        # Read data from device
        l, data = inp.read()
      
        if l:
            f.write(data)
            time.sleep(0.001)

Thanks in advance.

ossilator, 2022-08-05:
is this reproducible with v0.9, using the new method to parameterize the pcm object?

if so, what does inp.dumpinfo() say?

RonaldAJ, 2022-11-14:
I think this is an ALSA problem. If the hardware does not support 48kHz then you won't get 48kHz  without defining a "virtual" device. pyalsaaudio is just a thin wrapper around ALSA. 

You need to define a "plugin" in the ALSA configuration. It is best to read up on the documentation here: https://www.alsa-project.org/main/index.php/Asoundrc#Why_asoundrc.3F. And then ask questions in the ALSA forum.

ossilator, 2024-02-01:
reporter seems to have lost interest.

closing as "presumably user error".

closed 2024-02-01 by ossilator