larsimmisch/pyalsaaudio #68: alsaaudio.Mixer() fails on Raspberry Pi because there is no Master channel

ssokolow, 2018-12-18:
After using `pip3 install pyalsaaudio` (which gave me version 0.8.4 when I ran it on 2018-12-17), attempting to instantiate `Mixer` raises an exception because the Raspberry Pi's mixer has only a single `PCM` channel but no `Master` channel.

```
pi@raspberrypi:~ $ python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import alsaaudio
>>> alsaaudio.Mixer()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
alsaaudio.ALSAAudioError: Unable to find mixer control Master,0 [default]
>>>
pi@raspberrypi:~ $ amixer
Simple mixer control 'PCM',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback 400 [100%] [4.00dB] [on]
pi@raspberrypi:~ $
```

As `help(alsaaudio.Mixer)` didn't say anything about alternative arguments to the constructor, I resorted to using `subprocess.check_call` with the `amixer` command instead.

hein0r, 2019-04-12:
Upgrading to the newest version of pyalsaaudio solved this issue for me

GhastSlayer2907, 2019-04-27:
Just use alsaaudio.Mixer('PCM')

ssokolow, 2019-04-27:
@GhastSlayer2907 

You expect end users to patch software they had no hand in writing or nag programmers who have no Pi to test on, just because the Pi defies a convention every other platform follows?

GhastSlayer2907, 2019-04-27:
Hey man chill, I wasn't trying to "nag" you or anything, just trying to help. I'm sorry if my answer came across as unhelpful. I'm just 13 bro

ssokolow, 2019-04-27:
Ahh, sorry about that. It was also my mistake. I try to be polite and assume the best possible intentions but, every now and then, when I'm feeling grumpy, I accidentally treat people as if they know how many other unhelpful things I've seen or heard recently.

I should have just said something like this:

> Thanks for the effort, but I think I tried that and it didn't work with the version of pyalsaaudio I was using at the time.
> 
> Either way, it's still something that should be fixed, since not everyone who tries to run something pyalsaaudio-based on a Pi knows how to edit that particular program.

GhastSlayer2907, 2019-04-27:
It's ok. Happens to me too, a lot :-)

miccoli, 2019-08-18:
> After using pip3 install pyalsaaudio (which gave me version 0.8.4 when I ran it on 2018-12-17), attempting to instantiate Mixer raises an exception because the Raspberry Pi's mixer has only a single PCM channel but no Master channel.

@ssokolow please note that the blame here is only on the guys that write the alsa drivers, where (apparently) there is no agreed upon convention on the names of mixer simple controls.

On my RPi, e.g. I have:

```
$ amixer scontrols
Simple mixer control 'DSP Program',0
Simple mixer control 'Analogue',0
Simple mixer control 'Analogue Playback Boost',0
Simple mixer control 'Auto Mute',0
Simple mixer control 'Auto Mute Mono',0
Simple mixer control 'Auto Mute Time Left',0
Simple mixer control 'Auto Mute Time Right',0
Simple mixer control 'Clock Missing Period',0
Simple mixer control 'Deemphasis',0
Simple mixer control 'Digital',0
Simple mixer control 'Max Overclock DAC',0
Simple mixer control 'Max Overclock DSP',0
Simple mixer control 'Max Overclock PLL',0
Simple mixer control 'Volume Ramp Down Emergency Rate',0
Simple mixer control 'Volume Ramp Down Emergency Step',0
Simple mixer control 'Volume Ramp Down Rate',0
Simple mixer control 'Volume Ramp Down Step',0
Simple mixer control 'Volume Ramp Up Rate',0
Simple mixer control 'Volume Ramp Up Step',0
```

So no `Master`, no `PCM`... BTW if you are looking to set volume, you have to use `alsaaudio.Mixer('Digital')`

Given the horrible state of the alsa drivers and alsa docs, I think there is really no possibility to guess a sensible name for the 'default' alsa simple mixer control on every possible platform and configuration.

IMHO the only way to fix this issue is either
- make the `control` option of `Mixer` mandatory, and remove the `control='Master'` default
- add to the docs a warning that the alsa driver authors are free to chose the name of the 'default' mixer scontrol to whatever weird name they like, so no hope that `alsaaudio.Mixer()` picks a sensible default in every case.

PS:

@GhastSlayer2907 comment is appropriate and helpful to me, and other users I think. It simply states the fact that if the 'default' mixer scontrol does not exist, you have to explicitly choose the appropriate one.

> I can't remember clearly, but I think I did try alsaaudio.Mixer('PCM') and my Pi was running a version of pyalsaaudio which gave me an error about the wrong number of arguments.

I can confirm that on v. `0.8.4` the solution proposed by @GhastSlayer2907 works like a charm, if, of course, you choose the appropriate scontrol name.

ssokolow, 2019-08-18:
> please note that the blame here is only on the guys that write the alsa drivers, where (apparently) there is no agreed upon convention on the names of mixer simple controls.

Ugh. I suspect that `Analogue` in your listing is what gets named `PCM` on both my Pi and all the desktop PCs I've ever owned.

(My desktop PC has the motherboard audio having `Master`, `PCM`, and `Digital`, with `PCM` being the analog speaker outputs and Digital being the S/PDIF. Then the HDMI digital output is a completely separate device with its own mixer.)

miccoli, 2019-08-18:
The driver for my card is quite different.

'Digital' is the actual (you guess, digital) playback volume control, with a range from -103.0  to 0.00 dB, in  0.5 dB steps.  

'Analogue' has just two values, 0 which corresponds to -6.0 dB attenuation, and 1, for 0.00 dB attenuation. The name suggests that this attenuation is obtained on the 'analog' side of the DAC, so it is just a switch to match the output of the DAC to the sensitivity of the amplifier, without "touching" the digital side of the DA conversion.

Apparently this is an old (now deprecated?) naming convention that somehow survived in the driver my card uses, see <https://github.com/raspberrypi/linux/blob/73a743bcf8618982afab993a25cc36df0d9d9ae4/sound/soc/codecs/pcm512x.c#L446-L449>

Bottom line: the only way to get things working is a lot of experimentation, and for now implementing heuristics in a libaudio wrapper is way to complicated.

Tom4259, 2020-06-08:
Hi,
I fixed this problem by doing:
`alsamixer`
then i got the name from the Item at the top left and put it in the:
`alsaaudio.Mixer('<YOUR` ITEM>')`
Tested it out, it worked!

ssokolow, 2020-06-08:
@Tom4259 People have said that before.

The reason this is a bug is that programs which work everywhere else break on the Raspberry Pi, and there's no guarantee a user will know how to poke around inside someone else's code (which could be a real mess) to find the line that needs to be changed.

larsimmisch, 2020-07-16:
@ssokolow This can not be meaningfully fixed in pyalsaaudio or even in ALSA at this point in time.

ALSA is a framework that ties an API to a Driver level Service Provider Interface and the lack of examples and clear guidance have contributed to competing conventions. Trying to unify those by guessing what a user might want will ultimately make matters more complicated.

closed 2020-07-16 by larsimmisch