larsimmisch/pyalsaaudio #69: Setting volume to 0 using Mixer.setvolume doesn't set gain to -inf

ssrselvamraju, 2019-01-17:
Setting volume to 0 using:

```
mixer_args = {'cardindex': 1, 'device': 'pulse'}
alsaaudio.Mixer(**mixer_args).setvolume(0)
```
doesn't set gain to infinity on the card (on some computers you actually hear the very mild audio). Alsamixer shows Master volume as 0 though.

```
$ pactl list sinks | grep -e "Sink" -e Volume
Sink #1
	Volume: front-left: 5358 /   8% / -65.25 dB,   front-right: 5358 /   8% / -65.25 dB
	Base Volume: 65536 / 100% / 0.00 dB
```

Setting volume to 0 on command line (specifying the device as pulse), sets the volume gain to -inf (truly sets all channels on alsamixer to 0).

```
$ amixer -D pulse sset Master 0
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 0 [0%] [on]
  Front Right: Playback 0 [0%] [on]

$ pactl list sinks | grep -e "Sink" -e Volume
Sink #0
	Volume: front-left: 87248 / 133% / 7.46 dB,   front-right: 87248 / 133% / 7.46 dB
	Base Volume: 65536 / 100% / 0.00 dB
Sink #1
	Volume: front-left: 0 /   0% / -inf dB,   front-right: 0 /   0% / -inf dB
	Base Volume: 65536 / 100% / 0.00 dB
```
P.S. not specifying `-D pulse` on command line has the same effect as the issue above.

stalkerg, 2020-05-21:
For that case usually using the mute.

ossilator, 2024-02-01:
as documented, cardindex overrides device. so you presumably end up addressing the actual hardware device, the same as amixer without -D pulse, so you get consistent results.

the exact meaning of minimal volume is driver-dependent.

pulse's virtual device amalgamates the volume with mute, so you get a different result there.

closed 2024-02-01 by ossilator