larsimmisch/pyalsaaudio #70: Read() hangs in Raspbian randomly

kishorkl, 2019-02-25:
I have a USB 4 channel mic and am trying to capture audio at 16000Hz using pyalsaaudio 0.8.4 in a raspberry pi 3 B+. I run the audio capture in a thread, which can be stopped if user inputs -1, which is detected using another thread.  The wave files are limited to 3 minutes to prevent creating large wave files. This works most of the time but, randomly the audio capture thread hangs in the read call, as seen through a pyrasite probe (https://gist.github.com/reywood/e221c4061bbf2eccea885c9b2e4ef496). 

I have attached the code I used with Python 3.5.3 with Raspbian Linux 9.6.
[alsatest.zip](https://github.com/larsimmisch/pyalsaaudio/files/2900812/alsatest.zip)

Sometimes the hang happens within 45 seconds of recording, sometimes after 20 minutes of recording.

Could you help me in finding what I am doing wrong or if there is a bug in pyalsaaudio?

Thanks,
Kishor

kishorkl, 2019-02-25:
Additionally, if I increase the duration threshold to 10 minutes (instead of 3 minutes as in code), the hang happens within the third wav file capture (20+ minutes) for sure.

Thanks,
Kishor

kishorkl, 2019-02-25:
[alsatest.zip](https://github.com/larsimmisch/pyalsaaudio/files/2900868/alsatest.zip)

Use the attached source code, where I corrected some of the comments. Sorry for repeated posting.

Thanks,
Kishor

ossilator, 2022-08-05:
most likely a driver problem; upgrading the kernel might work.
not much to be done about it here.

closed 2024-02-01 by ossilator