larsimmisch/pyalsaaudio #75: Playing music and changing the volume at the same time in the same program

katoabc123, 2019-09-23:
I would like to write a python program to play the music and increase the volume in the same program.

However, I encounter a problem: 
when the music is playing, the program is holding at device.write(data) line.So, only after finishing the music, the set_volume function run. Now I would like to do them at the same time. Do you have any suggestion? I don't want to create a new python file to separate the play and set_volume line.

For example:
play(music_device, wav_file)
time.sleep(5)
mixer.setvolume(volume, channel))

larsimmisch, 2020-07-13:
Well, there should still be an order, no? So you might want to change the volume before the music plays, or while it plays, and I think you can do that. Just not by using a `play` function that plays an entire file, but you could do it inside the inner loop in `play`.

ossilator, 2022-08-05:
close?

closed 2024-02-01 by ossilator