larsimmisch/pyalsaaudio #76: Recording at 384,000 Hz

paddygoat, 2019-10-10:
I can record at a max of 192,000 Hz, but need to record at 384,000. Any ideas?

`Fs = 192000
#Fs = 384000

def recordAudioSegments(RecordPath, BLOCKSIZE):	
    # This function is used for recording audio segments (until ctr+c is pressed)
    # ARGUMENTS:
    # - RecordPath:		the path where the wav segments will be stored
    # - BLOCKSIZE:		segment recording size (in seconds)
    # 
    # NOTE: filenames are based on clock() value

    print ("Press Ctr+C to stop recording")
    RecordPath += os.sep
    d = os.path.dirname(RecordPath)
    if os.path.exists(d) and RecordPath!=".":
        shutil.rmtree(RecordPath)	
    os.makedirs(RecordPath)	

    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE,alsaaudio.PCM_NONBLOCK)
    inp.setchannels(2)
    inp.setrate(Fs)
    inp.setformat(alsaaudio.PCM_FORMAT_S16_LE)
    inp.setperiodsize(512)
    midTermBufferSize = int(Fs*BLOCKSIZE)
    midTermBuffer = []
    curWindow = []
    elapsedTime = "%08.3f" % (time.time())
    while 1:
        l,data = inp.read()
        if l:
            for i in range(len(data)/2):
                curWindow.append(audioop.getsample(data, 2, i))

            if (len(curWindow)+len(midTermBuffer)>midTermBufferSize):
                samplesToCopyToMidBuffer = midTermBufferSize - len(midTermBuffer)
            else:
                samplesToCopyToMidBuffer = len(curWindow)

            midTermBuffer = midTermBuffer + curWindow[0:samplesToCopyToMidBuffer];
            del(curWindow[0:samplesToCopyToMidBuffer])

        if len(midTermBuffer) == midTermBufferSize:
            # allData = allData + midTermBuffer				
            curWavFileName = RecordPath + os.sep + str(elapsedTime) + ".wav"				
            midTermBufferArray = numpy.int16(midTermBuffer)
            wavfile.write(curWavFileName, Fs, midTermBufferArray)
            print ("AUDIO  OUTPUT: Saved ") + curWavFileName
            midTermBuffer = []
            elapsedTime = "%08.3f" % (time.time())	`

larsimmisch, 2019-10-11:
I’d be surprised if the underlying hardware supported that. Are you sure it does?

paddygoat, 2019-10-11:
Thanks for the reply @larsimmisch larsimmisch. I'm using a [Dodotronics UltraMic384K](https://www.dodotronic.com/product/ultramic-384k-ble/?v=79cba1185463) on a new mid range Lenovo laptop with Ubuntu 18 and have been recording successfully using Audacity at 384K. Audacity 'says' it's using Alsa, but I've never managed to get Alsa working beyond 192 using command line. I'm totally stuck!  Any thoughts? ....... Thanks.

larsimmisch, 2019-10-11:
Very cool. No idea, sorry. If Audacity manages to do it, I‘d read the Audacity source code.

closed 2020-04-05 by larsimmisch