larsimmisch/pyalsaaudio #77: Unplugging a USB DAC alsaaudio.PCM operates on freezes the system.

crocket, 2019-10-19:
Whenever I play audio, opening the USB DAC takes 500 milliseconds. Thus, I use the following program to keep my USB DAC open.

```
#!/usr/bin/env python
import alsaaudio
import time
import sys

if __name__ == "__main__":
  device=alsaaudio.PCM(device=sys.argv[1])
  while True:
    time.sleep(1)
```

If I unplug the USB DAC while the program is running, the program becomes an immortal zombie process that cannot be killed. My system freezes within a few minutes. Keyboard and mouse stop working very soon.

larsimmisch, 2019-10-21:
I'm afraid there is nothing pyalsaaudio can do there. It installs no signal handlers. As a general rule of thumb, a user-level binary should not be able to crash a system - the problem is probably a faulty driver.

closed 2019-10-21 by larsimmisch

crocket, 2019-10-21:
You mean snd-usb-audio.