larsimmisch/pyalsaaudio #78: close() hangs after non-blocking PCM capture

bigzed, 2019-11-16:
if I modify the included 'recordtest.py' to record sound in a function, as shown below, the function never returns and 'inp.close()' hangs and it utilizes 100% CPU until SIGTERM. See the print statements marked with 'NEVER REACHED'.
 
```python
## To test it out do the following:
## python recordtest.py out.raw # talk to the microphone
## aplay -r 8000 -f S16_LE -c 1 out.raw

#!/usr/bin/env python

from __future__ import print_function

import sys
import time
import getopt
import alsaaudio

def usage():
    print('usage: recordtest.py [-d <device>] <file>', file=sys.stderr)
    sys.exit(2)

def record(file):
    f = open(file, 'wb')

    # Open the device in nonblocking capture mode. The last argument could
    # just as well have been zero for blocking mode. Then we could have
    # left out the sleep call in the bottom of the loop
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NONBLOCK, device=device)

    # Set attributes: Mono, 44100 Hz, 16 bit little endian samples
    inp.setchannels(1)
    inp.setrate(8000)
    inp.setformat(alsaaudio.PCM_FORMAT_U8)

    # The period size controls the internal number of frames per period.
    # The significance of this parameter is documented in the ALSA api.
    # For our purposes, it is suficcient to know that reads from the device
    # will return this many frames. Each frame being 2 bytes long.
    # This means that the reads below will return either 320 bytes of data
    # or 0 bytes of data. The latter is possible because we are in nonblocking
    # mode.
    inp.setperiodsize(160)

    loops = 1000000
    print('start recording')
    while loops > 0:
        loops -= 1
        # Read data from device
        l, data = inp.read()

        if l:
            f.write(data)
            time.sleep(.001)
    print('finished')
    inp.close()
    print('closed') # NEVER REACHED

if __name__ == '__main__':
    device = 'default'

    opts, args = getopt.getopt(sys.argv[1:], 'd:')
    for o, a in opts:
        if o == '-d':
            device = a

    if not args:
        usage()

    record(args[0])
    print('done') # NEVER REACHED
```

daym, 2019-12-02:
Yeah, we debugged it and it's easy to fix by calling the pause function before closing.

What is happening is that PulseAudio is handling the hardware side and providing "a" pipe to clients like your application. When closing, PulseAudio client library tries to drain the pipe in order to prevent a broken pipe error. This will not work as long as capture is enabled since the universe doesn't stop providing the signal.

larsimmisch, 2020-04-02:
#80 is merged on master - closing

closed 2020-04-02 by larsimmisch