larsimmisch/pyalsaaudio #81: Wrong data in background thread

hifiberry, 2020-01-23:
I'm using a quite simple "alsaloop" implementation that just copies input to output. It has been derived from the sample play and record programs to make sure I didn't mess up something ;-)

    def simple_loop(self):
        device = 'default'

        # Open the device in nonblocking capture mode. The last argument could
        # just as well have been zero for blocking mode. Then we could have
        # left out the sleep call in the bottom of the loop
        inp = PCM(PCM_CAPTURE, PCM_NONBLOCK, device=device)
    
        # Set attributes: Stereo, 44100 Hz, 16 bit little endian samples
        inp.setchannels(2)
        inp.setrate(44100)
        inp.setformat(PCM_FORMAT_S16_LE)
    
    
        # Open the device in playback mode.
        out = PCM(PCM_PLAYBACK, device=device)
    
        # Set attributes:Stereo, 44100 Hz, 16 bit little endian frames
        out.setchannels(2)
        out.setrate(44100)
        out.setformat(PCM_FORMAT_S16_LE)
    
        # The period size controls the internal number of frames per period.
        # The significance of this parameter is documented in the ALSA api.
        out.setperiodsize(1024)
    
    
        # The period size controls the internal number of frames per period.
        # The significance of this parameter is documented in the ALSA api.
        # For our purposes, it is suficcient to know that reads from the device
        # will return this many frames. Each frame being 2 bytes long.
        # This means that the reads below will return either 320 bytes of data
        # or 0 bytes of data. The latter is possible because we are in nonblocking
        # mode.
        inp.setperiodsize(1024)
    
        loops = 1000000
        while loops > 0:
            loops -= 1
            # Read data from device
            l, data = inp.read()
            
            if (len(data) % 4) != 0:
                    logging.error("oops %s",len(data))
    
            if l:
                print(l)
                out.write(data)
                time.sleep(.001)

This code work well when running as the main thread. However, when I run this in a background thread (application uses GLib that needs to be the main thread), it doesn't work reliably. After a few second, it breaks. It not just doesn't send data to the output anymore. The read data are also kind of incorrect as the length of the buffer sometimes isn't a multiple of the sample size.

Are there any know threading issues?

larsimmisch, 2020-01-25:
To answer the direct question: no and yes. No, there are no known threading issues in that I'm relatively sure that the GIL handling of pyAlsaAudio is correct, and yes in the sense that it's still Python code and GIL side-effects can be subtle.

I guess I'd look into integrating alsaaudio into the main GLib event loop using `polldescriptors` and writing single-threaded, reactive code instead of spawning a thread. But I know nothing about GLib.

P.S. If this means you're working on your new card with DAC and ADC, yay!

hifiberry, 2020-01-25:
I guess, I will probably just spawn an external process as I do it now with alsaloop as I only need to send a few byte for control information between the two processes stdin/stdout will do for this.

P.S. Yes, I was trying to replace alsaloop and make it a bit more flexible (to loop only if there is a signal on the input). But it will have to wait now for the next release ;-)

closed 2020-04-02 by larsimmisch