larsimmisch/pyalsaaudio #95: Event when a sound is being played

peter-vanpoucke, 2020-12-06:
Couldn't really find in the documentation if it's possible to raise an event when somethings being played. By that I mean if any proces was playing a sound. Could also be interesting to see if something's recording too.

I guess it's out of scope for this library to cater for that, but I just wanted to raise the question. For now it seems going over the card's sub status in `/proc` seems to be the only way.

larsimmisch, 2020-12-06:
That's a really interesting question (and I have no idea how to do that more directly, sorry). I'll leave the issue open and have added the help wanted label, maybe someone else knows.

ossilator, 2022-08-05:
peter, what do you need this for?

i don't think alsa provides such functionality. some sound servers might provide hooks into which you could wire notifications.

generalized kernel tracing facilities might also provide such a function, but you'd need to filter out the noise.

in any case, out of scope here.

closed 2024-02-01 by ossilator