larsimmisch/pyalsaaudio #96: PyAlsaAudio refuses to compile with errors

DanRPI, 2021-01-02:
i tried to install pyalsaaudio with pip3 (to get another script running)
(as root on a RPI1, Raspbian GNU/Linux 8 (jessie) 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux)
# pip3 install pyalsaaudio

Downloading/unpacking pyalsaaudio
  Downloading pyalsaaudio-0.9.0.tar.gz (370kB): 370kB downloaded
  Running setup.py (path:/tmp/pip-build-qunox10l/pyalsaaudio/setup.py) egg_info for package pyalsaaudio

    warning: no files found matching '*.gif' under directory 'doc'
    warning: no files found matching '*.json' under directory 'doc'
Installing collected packages: pyalsaaudio
  Running setup.py install for pyalsaaudio
    building 'alsaaudio' extension
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.4m -c alsaaudio.c -o build/temp.linux-armv6l-3.4/alsaaudio.o
    alsaaudio.c: In function 'alsapcm_getformats':
    alsaaudio.c:663:2: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
      for (size_t i = 0; i < ARRAY_SIZE(ALSAFormats); ++i) {
      ^
    alsaaudio.c:663:2: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
    alsaaudio.c: In function 'alsapcm_getrates':
    alsaaudio.c:754:3: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
       for (size_t i=0; i<ARRAY_SIZE(ALSARates); i++) {
       ^
    alsaaudio.c: In function 'alsapcm_getchannels':
    alsaaudio.c:805:2: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
      for (unsigned ch=min;ch<=max;++ch) {
      ^
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    Complete output from command /usr/bin/python3 -c "import setuptools, tokenize;__file__='/tmp/pip-build-qunox10l/pyalsaaudio/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-bx90n2rn-record/install-record.txt --single-version-externally-managed --compile:
    running install

running build

running build_ext

building 'alsaaudio' extension

creating build

creating build/temp.linux-armv6l-3.4

arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.4m -c alsaaudio.c -o build/temp.linux-armv6l-3.4/alsaaudio.o

alsaaudio.c: In function 'alsapcm_getformats':

alsaaudio.c:663:2: error: 'for' loop initial declarations are only allowed in C99 or C11 mode

  for (size_t i = 0; i < ARRAY_SIZE(ALSAFormats); ++i) {

  ^

alsaaudio.c:663:2: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code

alsaaudio.c: In function 'alsapcm_getrates':

alsaaudio.c:754:3: error: 'for' loop initial declarations are only allowed in C99 or C11 mode

   for (size_t i=0; i<ARRAY_SIZE(ALSARates); i++) {

   ^

alsaaudio.c: In function 'alsapcm_getchannels':

alsaaudio.c:805:2: error: 'for' loop initial declarations are only allowed in C99 or C11 mode

  for (unsigned ch=min;ch<=max;++ch) {

  ^

error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

----------------------------------------
Cleaning up...
Command /usr/bin/python3 -c "import setuptools, tokenize;__file__='/tmp/pip-build-qunox10l/pyalsaaudio/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-bx90n2rn-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-qunox10l/pyalsaaudio
Storing debug log for failure in /root/.pip/pip.log

i did the same for pyaudio which worked after i installed libportaudio2 and portaudio19-dev
so it doesn't seem to be due to pip3 or phyton itself (which i both installed to get...........)
i really don't like this system bloating loops one has to make to get things working...
in 2k one could start a desktop computer with a 350MB Linux, now one needs 2gig for a stripped version minibian for a single platine rsc computer...

DanRPI, 2021-01-03:
got it working 
a "md m4" is missing in the autoconfig

ABesginow, 2021-02-16:
> got it working
> a "md m4" is missing in the autoconfig

Can you maybe provide a bit more information on your solution?
I am encountering the same error message while installing on an RPi3 with dietpi OS.

DanRPI, 2021-02-17:
a little more detail as requested, from memory, it was some weeks ago, in plain speak, i not that expert...
- during the depacking script of the modules i noticed an error reading the like "folder m4 is missing ... extracting to" and a while after these files were missing in the compile run.
- so i simply made this folder at the expected location, 
- and in the next run the files went there while extraction and were finally found in the compile run...

jiansongyang, 2021-03-11:
As a work around it may help to specify c99 as the C++ specification. So, in the file setup.py:
` ext_modules=[Extension('alsaaudio',['alsaaudio.c'],
                               libraries=['asound'],
                               extra_compile_args=["-std=c99",]
        )]
`

jiansongyang, 2021-03-11:
> i tried to install pyalsaaudio with pip3 (to get another script running)
> (as root on a RPI1, Raspbian GNU/Linux 8 (jessie) 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux)
>
> # pip3 [...]

As a work around it may help to specify c99 as the C++ specification. So, in the file setup.py:

`ext_modules=[Extension('alsaaudio',['alsaaudio.c'],
                               libraries=['asound'],
                               extra_compile_args=["-std=c99",]
        )]`