larsimmisch/pyalsaaudio #102: Code for detecting electromagnetic crosstalk

RonaldAJ, 2021-05-04:
Dear Lars,

I have written scripts to record sound which will show electromagnetic crosstalk from Wi-Fi or the recording electronics to the microphone. It will be accompanied by a short note (paper would be too much honour for this piece of text). For an early preview: https://github.com/soundappraisal/pyalsaaudio/tree/Establishing_electro_magnetic_crosstalk. 

Is it something you would consider to include in the library, or is it better to give it its own repo?

Kind regards,
Ronald

larsimmisch, 2021-05-04:
Hi Ronald,

I don't have a strong opinion, TBH. But for the purposes of your paper, a separate repo (maybe with a pyalsaaudio submodule for easier installation) might be better, no?

- Lars

ossilator, 2022-08-05:
this seems way too high-level.
i suppose one could link it somewhere in the documentation, but there is *a lot* of other code one could link as well.
so, out of scope imo.

RonaldAJ, 2022-11-12:
I'll move it elsewhere.

closed 2022-11-12 by RonaldAJ