larsimmisch/pyalsaaudio #107: Selecting USB Audio Mic/Record volume

JediMongoose, 2022-01-16:
I'm using pyalsaaudio on a Raspberry Pi with a USB audio device. I want to read and set the volume of the "Mic" input.  With the amixer command, I can read the volume with:
`amixer -c 1 cget iface=MIXER,name='Mic Capture Volume`
And set it with:
`amixer -c 1 cset iface=MIXER,name='Mic Capture Volume' 50%`

So I try replicating this behaviour with the mixertest.py script, but no amount of changing Mixer names etc will let me access the Mic volume.

I've tried:
```
import alsaaudio
mixer = alsaaudio.Mixer("Capture")
volumes=mixer.getvolume()
print volumes
```
But this just returns an empty array :(

Any help greatly appreciated

```
pi@pi4a:~ $ python mixertest.py Master
Available sound cards:
  1: USB Audio Device (GeneralPlus USB Audio Device at usb-0000:01:00.0-1.1, full speed)
Mixer name: 'Master'
Capabilities: Playback Volume Playback Mute Joined Playback Mute
Channel 0 volume: 22%
Channel 1 volume: 22%
Channel 0 is muted
Channel 1 is muted
pi@pi4a:~ $ python mixertest.py Capture
Available sound cards:
  1: USB Audio Device (GeneralPlus USB Audio Device at usb-0000:01:00.0-1.1, full speed)
Mixer name: 'Capture'
Capabilities: Capture Volume Capture Mute Joined Capture Mute
Channel 0 is recording
Channel 1 is recording
```

```
pi@pi4a:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
default
    Playback/recording through the PulseAudio sound server
sysdefault:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Hardware device with all software conversions
usbstream:CARD=Device
    USB Audio Device
    USB Stream Output
```

larsimmisch, 2022-01-16:
Try creating a mixer like this: `Mixer(cardindex=1)`

JediMongoose, 2022-01-17:
Hi. Thanks for the tip. I tried what you suggested, but get this:
```
Python 2.7.16 (default, Oct 10 2019, 22:02:15) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import alsaaudio
>>> mixer=alsaaudio.Mixer(cardindex=1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
alsaaudio.ALSAAudioError: Unable to find mixer control Master,0 [hw:1]
```

I also tried:
```
>>> mixer=alsaaudio.Mixer("Mic Capture Volume",cardindex=1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
alsaaudio.ALSAAudioError: Unable to find mixer control Mic Capture Volume,0 [hw:1]

>>> mixer=alsaaudio.Mixer("Mic",cardindex=1)
>>> volumes=mixer.getvolume()
>>> print volumes
[50]
```

which I thought was working!  But, when I check with alsamixer, it's the 'wrong' Mic - I need the Capture one, not the Playback one.

Just to clarify, this command does show the volume from the 'correct' Mic:
```
$ amixer -c 1 cget iface=MIXER,name='Mic Capture Volume'
numid=8,iface=MIXER,name='Mic Capture Volume'
  ; type=INTEGER,access=rw---R--,values=1,min=0,max=30,step=0
  : values=19
  | dBminmax-min=-12.00dB,max=33.00dB
```