larsimmisch/pyalsaaudio #110: Comment on default periodsize

drmikecooke, 2022-03-19:
The documentation says that the default is 32. This is only true for 16-bit stereo samples (4bytes in total) giving 128 bytes. For 8bit mono you get back 128 bytes, not the 32 I expected. I then included periodsize in my declaration (1024), and for 8bit mono I got back 1024 bytes, as I would expect. Doesn't particularly matter to what I am doing, but it's a loose end.

larsimmisch, 2022-03-20:
If you have an idea how to put this into words in the documentation and create a PR, I'd merge it right away :)

RonaldAJ, 2022-08-26:
I think it is necessary to check whether you got what you asked for. The ALSA method used for setting period_size (snd_pcm_hw_params_set_period_size_near) does not guarantee that you get the specified number of frames, instead the documentation states: "Restrict a configuration space to have period size nearest to a target. "

After instantiating the PCM device you can check the periodsize by calling the info method on the PCM device. Some fields in the returned structure are prefixed with '(call value)'. These contain the values you set on the PCM device even if propagation to ALSA was unsuccessful. The periodsize is obtained directly from ALSA, so should reflect the current status of the PCM device.

In my current setup I get 64 back when I try 32, 64, 75 or 1024 as periodsize during instantation.

closed 2024-02-01 by ossilator