larsimmisch/pyalsaaudio #111: Help for dummp needed

StuartIanNaylor, 2022-04-02:
My mixer is 

```
Simple mixer control 'Speaker',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 50
  Mono:
  Front Left: Playback 30 [60%] [-20.00dB] [on]
  Front Right: Playback 30 [60%] [-20.00dB] [on]
Simple mixer control 'Mic',0
  Capabilities: pvolume pvolume-joined cvolume pswitch pswitch-joined cswitch cswitch-joined
  Playback channels: Mono
  Capture channels: Front Left - Front Right
  Limits: Playback 0 - 20 Capture 0 - 20
  Mono: Playback 16 [80%] [24.00dB] [off]
  Front Left: Capture 20 [100%] [30.00dB] [on]
  Front Right: Capture 20 [100%] [30.00dB] [on]
Simple mixer control 'Auto Gain Control',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
```
card is c1 but I can not work out how to set the volume for mic even the example errors

```
./mixertest.py -c1 Mic
Available sound cards:
  0: vc4-hdmi (vc4-hdmi)
  1: VIA USB Dongle (VIA Technologies Inc. VIA USB Dongle at usb-3f980000.usb-1.4, full speed)
Mixer name: 'Mic'
Capabilities: Playback Volume Joined Playback Volume Capture Volume Playback Mute Joined Playback Mute Capture Mute Joined Capture Mute
Traceback (most recent call last):
  File "/home/pi/pyalsaaudio/./mixertest.py", line 152, in <module>
    show_mixer(args[0], kwargs)
  File "/home/pi/pyalsaaudio/./mixertest.py", line 52, in show_mixer
    pmin_keyword, pmax_keyword = mixer.getrange(pcmtype=alsaaudio.PCM_PLAYBACK, units=alsaaudio.VOLUME_UNITS_RAW)
AttributeError: module 'alsaaudio' has no attribute 'VOLUME_UNITS_RAW'
```
If you know the card and Mixer whats the easiest and most direct way to control?

:) apols but its just not computing today or am I missing something.
Any tips would be greatly appreciated

larsimmisch, 2022-04-02:
`VOLUME_UNITS_RAW` is from a fairly recent PR and you are getting an `AttributeError: module 'alsaaudio' has no attribute 'VOLUME_UNITS_RAW'`. I suspect that you have an older version of `pyalsaaudio` that gets picked up by your environment.

If you want to see where it comes from, try:
```
$ python3 -v
>>> import alsaaudio
```

StuartIanNaylor, 2022-04-03:
Never used pyalsaaudio before and just installed into a venv fresh so doubt there was.
Debian 11
The softvol never worked as expected anyway as the changes to volume seemed to only work when next run.
So its all deleted but it was an absolute fresh install. Its alsa 1.2.4 on debian 11

closed 2022-04-03 by StuartIanNaylor