larsimmisch/pyalsaaudio #117: automatically using a wrong samplerate when the correct samplerate is unavailable

dev-jam, 2022-10-20:
This is basically a bug and a remark about a removed feature in one. 

Bug: when I open a device with a sample rate that is not supported by the soundcard, it will use another sample rate. For example, i have onboard audio which does not support 88200, when I select 88200 it will set it to 96000 and will play the audio around 10% faster (blocking mode). I would expect it to give an error with feedback like "sample rate of 88200 is not supported by the card". To workaround this bug, in the past I used to the deprecated method:

device = alsaaudio.PCM(type=alsaaudio.PCM_PLAYBACK, device=device_name)

                channels_set = device.setchannels(channels)
                samplerate_set = device.setrate(samplerate)
                period_size_set = device.setperiodsize(period_size)

and then compare the target values with the actual set values. If they don't agree then it will give an error. This deprecated method gave me some problems with the latest version so i decided to prepare for the future and phase out these deprecated methods. I now extract the actual set values from the shell with a command like this (different for each soundcard):

cat /proc/asound/PCH/pcm0p/sub0/hw_params

Is there a way to make the device give an error when it cannot set a value? Or are you planning to implement a new method to get the actual set values from the device?

Best,
Bob

larsimmisch, 2022-10-20:
There is a `PCM` method called `info` that returns a dictionary with - among other things - the `rate`. It's just undocumented, but I intend to document (and keep) it.

closed 2022-10-20 by larsimmisch

RonaldAJ, 2022-11-07:
@larsimmisch As I worked on the dictionary version I might have enough knowledge to draft the documentation for info. Would that help you, or you prefer to write it yourself?

larsimmisch, 2022-11-07:
@RonaldAJ Oh, I'd be delighted about a PR

RonaldAJ, 2022-11-07:
OK. I'll try to find time later this week.