larsimmisch/pyalsaaudio #124: Mixer.getvolume|setvolume percentage inconsistency?

rnixx, 2023-02-25:
Getting and setting volume on mixer by percentage not matches the percentage value displayed in `alsamixer`. Is this by intention and if so, why?

[image]

diederikdq, 2023-06-28:
Hi @rnixx ,
What percentage values do you get when you run:
```
amixer -M get Master
``` 
VS
```
amixer get Master
``` 
I had the same question about a month ago and found https://bbs.archlinux.org/viewtopic.php?pid=1114261#p1114261 
I can only share my findings when I tried to understand the different values.

ossilator, 2024-02-01:
the link provided above nails it: alsamixer uses a different scale than amixer and pyalsaaudio (logarithmic vs. linear).

closed 2024-02-01 by ossilator