larsimmisch/pyalsaaudio #125: unable to install hw params

FSet89, 2023-03-01:
I have the following error when trying to initialize PCM:
```
import alsaaudio as a
a.PCM(0)
```
**ALSA lib ../../../alsa-lib-1.1.9/src/pcm/pcm_direct.c:1271:(snd1_pcm_direct_initialize_slave) unable to install hw params
ALSA lib ../../../alsa-lib-1.1.9/src/pcm/pcm_dmix.c:1120:(snd_pcm_dmix_open) unable to initialize slave
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
alsaaudio.ALSAAudioError: Invalid argument [default]**

However, `arecord` on the same device works well.

RonaldAJ, 2023-03-01:
0 corresponds to PCM_PLAYBACK

Does your default device support playback?

FSet89, 2023-03-02:
> What happens if you specify an existing device instead of the implicit 'default':
> 
> `p=alsaaudio.PCM(1, device='null')`
> 
> Other solution might be to add default device to the ALSA configuration.

In this case I get no errors but I'm not sure this is the correct way:
```
import alsaaudio as a
p=a.PCM(1, device='null')
p.cardname()
```
**'null'**
`p.read()`
**(32, b'b\x00\x00\x00\x17\x00\x00\x00\x00\x00\x00\x00J\x00\x00\x00\x00\x00\x00\x004\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x003\x00\x00\x00>\x00\x00\x00?\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00G\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00L\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00F\x00\x00\x00Q\x00\x00\x008\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')**

RonaldAJ, 2023-03-02:
This looks like correct output. It needs to be converted from a byte string to a numerical format.

I have a conversion dictionary for that purpose:
`conversion_dict = {
            alsaaudio.PCM_FORMAT_S8: {'dtype': np.int8, 'endianness': '', 'formatchar': 'h', 'bytewidth': 1},
            alsaaudio.PCM_FORMAT_U8: {'dtype': np.uint8, 'endianness': '', 'formatchar': 'h', 'bytewidth': 1},
            alsaaudio.PCM_FORMAT_S16_LE: {'dtype': np.int16, 'endianness': '<', 'formatchar': 'h', 'bytewidth': 2},
            alsaaudio.PCM_FORMAT_S16_BE: {'dtype': np.int16, 'endianness': '>', 'formatchar': 'h', 'bytewidth': 2},
            alsaaudio.PCM_FORMAT_U16_LE: {'dtype': np.uint16, 'endianness': '<', 'formatchar': 'H', 'bytewidth': 2},
            alsaaudio.PCM_FORMAT_U16_BE: {'dtype': np.uint16, 'endianness': '>', 'formatchar': 'H', 'bytewidth': 2},
            alsaaudio.PCM_FORMAT_S32_LE: {'dtype': np.int32, 'endianness': '<', 'formatchar': 'l', 'bytewidth': 4},
            alsaaudio.PCM_FORMAT_S32_BE: {'dtype': np.int32, 'endianness': '>', 'formatchar': 'l', 'bytewidth': 4},
            alsaaudio.PCM_FORMAT_U32_LE: {'dtype': np.uint32, 'endianness': '<', 'formatchar': 'L', 'bytewidth': 4},
            alsaaudio.PCM_FORMAT_U32_BE: {'dtype': np.uint32, 'endianness': '>', 'formatchar': 'L', 'bytewidth': 4},
        }`

which is used with this function:

` def get_conversion_string(audioformat, noofsamples, conversion_dict ):
        conversion_string = f"{conversion_dict['endianness']}{noofsamples}{conversion_dict['formatchar']}"
        return conversion_string
`

to find the parameters for the conversion:
`data = np.array(struct.unpack(conversion_string, rawdata), dtype=dtype)`

Here rawdata is the second part returned by read, in your case:
`noofsamples, rawdata = p.read()`

These are just some edited cuts, so the code is not fully working. But should give you the right direction.

FSet89, 2023-03-02:
Thanks, I will give it a try. However, isn't 'null' the playback device? I'm expecting to see the card name when calling `p.cardname()`

RonaldAJ, 2023-03-02:
You can call p.info() to get information about the PCM device you opened.

RonaldAJ, 2023-03-05:
@FSet89 If it works you can close this issue.

closed 2024-02-01 by ossilator