larsimmisch/pyalsaaudio #128: Format of PCM Object

A622266, 2023-05-20:
I am having trouble setting the format of the PCM object.  I have version 0.10.0.
```python
device = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, format=PCM_FORMAT_S16_LE)
```
results in:
```
NameError: name 'PCM_FORMAT_S16_LE' is not defined
```
My code will run if I don't include the format parameter (i.e. accept the default).  But I want to change the default to other data types.  Does anyone have advice?

larsimmisch, 2023-05-20:
Oh, that’s just a missing `alsaaudio` namespace before  `PCM_FORMAT_S16_LE`. Try: `device = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, format=alsaaudio.  PCM_FORMAT_S16_LE)`

closed 2023-05-20 by larsimmisch

A622266, 2023-05-20:
yes thank you.  That was quite an oversight!