larsimmisch/pyalsaaudio #129: Broken Pipe

A622266, 2023-05-25:
Hi Lars:

In my application, I have a USB headset and an audio codec connected to a Raspberry PI3b.  The two PCM devices are configured as follows:
```python
    # Create the EVALADAU1761 sound card device
    evaladau1761_device = alsaaudio.PCM(alsaaudio.PCM_PLAYBACK, alsaaudio.PCM_NORMAL, format=alsaaudio.PCM_FORMAT_S16_LE, \
                                        rate=48000, channels=1, periodsize=8000, periods=4, cardindex=evaladau1761_card_index)
    # Create the USB headset device
    usb_headset_device = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, format=alsaaudio.PCM_FORMAT_S16_LE, \
                                       rate=48000, channels=1, periodsize=8000, periods=4, cardindex=usb_headset_card_index)
```
Data is captured from the headset mic and written to the audio codec in a continuous loop of a read from the headset followed by a write to the codec.
This runs for a minute or two, and while running, I can hear the output at the codec when speaking into the mic -- so it seems to be working fine.  But the program always ends in the following error:
```
alsaaudio.ALSAAudioError: Broken pipe [hw:1]
```
Can you offer some ideas on what may cause the broken pipe?  I think that if I had an overflow or underflow problem, I should just get bad sounding audio.  I believe that hw:1 is the codec, since the output of ```alsaaudio.cards()``` lists the codec as the second soundcard. I'm not sure how to approach debugging of the broken pipe.

Regards
Andy

larsimmisch, 2023-05-29:
Maybe this helps: https://stackoverflow.com/questions/65329417/alsa-broken-pipe-error-in-pcm-snd-writei-and-pcm-snd-readi

Generally, I think the way to go is to use nonblocking devices and the `poll` system call (or `select`/`epoll`), but I haven't managed to write an example yet.

A622266, 2023-06-01:
Thank you for your advice.   I'll look into the thread and the poll / epoll
calls.

ossilator, 2024-01-31:
duplicated by #130.

closed 2024-02-01 by ossilator