larsimmisch/pyalsaaudio #130: Broken Pipe in Version 0.10.0

Vishalghyv, 2023-06-30:
Code is working with package version 0.9.2 but not with the updated version.
I couldn't find any change log for the latest version nor any changes in commit log which mentioned changing format of declaration of PCM.

My code - 
```
device = 'default'
format = alsaaudio.PCM_FORMAT_FLOAT_LE
inp = alsaaudio.PCM(type=alsaaudio.PCM_PLAYBACK, mode = alsaaudio.PCM_NORMAL, channels=2, rate=8000, format=format,periods = 1,  periodsize=80, device = device)

```

Could you please help me here?

SchulerSimon, 2023-07-05:
I am having the same problem. 

At first I thought my code was the problem, but with 0.9.2 it works fine. 

My code tries to fix this by testing if writing to the pipe works. But after that first attempt was successful, it fails later. So at some point the pipe worked, from what I can tell. 

After some time, I get the same error as mentioned above: `alsaaudio.ALSAAudioError: Broken pipe [plug:airplay]`

For now, I will probably just go back to 0.9.2. 

```pyhton3
for _ in range(10):
    output_stream = alsaaudio.PCM(
        alsaaudio.PCM_PLAYBACK,
        alsaaudio.PCM_NORMAL,
        rate=self.rate,
        channels=self.channels,
        periodsize=int(self.frames_per_buffer / 2),
        device="plug:airplay",
    )
    # try to read audio, when failed continue next try
    try:
        raw_data = self.output_buffer.get(self.frames_per_buffer * 2)
        output_stream.write(raw_data) # WORKS HERE!
        self.output_stream_ready = True
        _LOGGER.debug("write to audio-buffer worked")
        break
    except alsaaudio.ALSAAudioError as e:
        _LOGGER.warning("could not read audio from alsa, retrying in 1 sec", exc_info=e)
        time.sleep(1)
        continue

while True: 
    raw_data = self.output_buffer.get(self.frames_per_buffer * 2)
    output_stream.write(raw_data) # FAILS HERE!
```

Metainfo: 
`lsb_release -a`
```
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal
```

`pip3 freeze | grep alsa`
```
pyalsaaudio==0.10.0
```

SchulerSimon, 2023-07-05:
Hi and thanks for the reply.

After some digging, I can give a bit more context: 

It only seems to appear when using input and output at the same time. The code below works for 0.9.2 (replays audio from mic to speaker) but not for 0.10.0. Strangely enough, it fails on output_stream.write(), but only when input_stream.read() data is provided. 

When playing silence (\x00 times frame_length) it works for both versions.  

```python
import alsaaudio

rate = 48000
channels = 1
format = alsaaudio.PCM_FORMAT_S16_LE
frame_length = 4096
input_device = "plug:user_mic_in"
output_device = "plug:airplay"

def audio_data_silence():
    while True:
        yield bytes([0] * frame_length)

input_stream = alsaaudio.PCM(
    alsaaudio.PCM_CAPTURE,
    alsaaudio.PCM_NORMAL,
    rate=rate,
    channels=channels,
    format=format,
    periodsize=int(frame_length / 2),
    device=input_device,
)

output_stream = alsaaudio.PCM(
    alsaaudio.PCM_PLAYBACK,
    alsaaudio.PCM_NORMAL,
    rate=rate,
    channels=channels,
    periodsize=int(frame_length / 2),
    device=output_device,
)

while True:
    # works for 0.9.2 but not 0.10.0
    _, raw_data = input_stream.read()
    output_stream.write(raw_data)

    # works for 0.9.2 and 0.10.0
    # output_stream.write(next(audio_data_silence()))
```

```
Traceback (most recent call last):
  File "alsaaudio_test.py", line 34, in <module>
    output_stream.write(raw_data)
alsaaudio.ALSAAudioError: Broken pipe [plug:airplay]
```

I wish you a great vacation.

larsimmisch, 2023-07-05:
I suspect it has to do with this commit: https://github.com/larsimmisch/pyalsaaudio/pull/123/commits/c2a6b6e58307034ad8febd3f7d296c9bd8727ca6. The old code was implicitly handling `-EPIPE` by calling `snd_pcm_recover` and then retrying. I suspect in the new version, the playback device is supposed to be re-opened when on underrun occurs. I'll play with it a bit, I can reproduce underruns when using pretty small buffers.

SchulerSimon, 2023-07-14:
Hello, thank you for the insight. I am not 100% sure I understand. 

Are you saying that it is supposed to be that way going forward, and that we should just deal with the error by reopening the output_device once the error occurs? Or is this going to be treated as error and addressed on your end?

larsimmisch, 2023-07-17:
I'll restore the previous behaviour for write

larsimmisch, 2023-08-09:
Can you take a look at https://github.com/larsimmisch/pyalsaaudio/pull/131 and let me know if it fixes the problem?

closed 2023-08-31 by larsimmisch

SchulerSimon, 2023-09-13:
Hey, thanks for addressing the issue. 

After I build and installed the current version of the code, it works just like in 0.9.2 with the given example code in my above comment.

larsimmisch, 2023-09-13:
Good to get confirmation, thanks!

ossilator, 2024-01-31:
i think it was wrong to restore the old behavior.

i failed to document it, and clearly wasn't thinking of that when i wrote the commit message, but the original motivation for the change was (probably) to stop papering over underruns, which are a serious enough error for a PCM stream.

in https://github.com/larsimmisch/pyalsaaudio/commit/946694d263d10b64a0fdad995f440f81bc245a40 i added the state() function, which could be examined for PCM_STATE_XRUN after catching the exception. the recovery from that would be simply calling write() again.

note that just calling state() prior to read()/write() is not adequate, as this introduces a race condition.

read() just returns the negative errno code on underrun. i guess this would be the right thing for write() to do as well - easy to ignore, and therefore sort of bug-to-bug compatible.

larsimmisch, 2024-01-31:
@ossilator I think I get it and in principle agree that explicit would be better than implicit here.

But: I restored the old behaviour (papering over underruns) because that's the way it was when I inherited this project (long ago). My speculation as to why: robust error handling for underruns is hard, so the original author (Casper Wilstrup) put it in to solve 80% of the typical problems with underruns.

It doesn't have to stay this way, but the other alternative would be:
- better documentation/better example code
- an announcement of a breaking change 

(and I have very little time ATM)

PR #131 (2023-08-31): Restore previous behaviour of calling snd_pcm_prepare in case of XRUN
I hope this will fix #130