larsimmisch/pyalsaaudio #135: [question] Control element "Master" not recognized in raw C Alsa library

jsgalarraga, 2023-09-04:
First of all, thanks for your work on this library and sorry for the ping here since the issue is not with the python library itself.

I'm working on C program that uses directly the ALSA library, and I'm trying to set the volume to a specific percentage value. I've tried to follow your implementation but I'm having one issue, when I set the element name to `"Master"` (as done [here](https://github.com/larsimmisch/pyalsaaudio/blob/main/alsaaudio.c#L1826)) I encounter the following error: `snd_mixer_selem_get_playback_volume_range: Assertion 'elem' failed`. However if I set element name to  `"PCM"` the volume range retrieval works fine, but I'm not able to set the percentage properly...

I'm working on a Raspberry Pi 4.

Also when I run `amixer scontrols` in the terminal the output is:
```
Simple mixer control 'Master',0
Simple mixer control 'Capture',0
```

Here's the code that I'm using:
```c
#include <stdio.h>
#include <alsa/asoundlib.h>

int main(int argc, char *argv[])
{
    long min, max;
    snd_mixer_t *handle;
    snd_mixer_selem_id_t *sid;
    const char *card = "default";
    const char *selem_name = "Master";

    snd_mixer_open(&handle, 0);
    snd_mixer_attach(handle, card);
    snd_mixer_selem_register(handle, NULL, NULL);
    snd_mixer_load(handle);

    snd_mixer_selem_id_alloca(&sid);
    snd_mixer_selem_id_set_index(sid, 0);
    snd_mixer_selem_id_set_name(sid, selem_name);
    snd_mixer_elem_t* elem = snd_mixer_find_selem(handle, sid);

    snd_mixer_selem_get_playback_volume_range(elem, &min, &max);

    printf("volume min: %ld - max: %ld\n", min, max);

}
```

I'm using the following command to compile it `gcc -o audio audio.c -lasound`

larsimmisch, 2023-09-05:
Hi,

> First of all, thanks for your work on this library and sorry for the ping here since the issue is not with the python library itself.
> I'm working on C program that uses directly the ALSA library, [...]

From what I remember, mixers default to setting/getting the raw volume. With amixer -c x scontents you should be able to see the limits.

What is a bit weird is that amixer shows a card which does have a ‘Master’ control, maybe there are multiple cards and you are selecting the wrong one in addition to th eproblem above?

Hope this helps,

- Lars

jsgalarraga, 2023-09-06:
I finally figured out my problem. I was executing my script with `sudo`:
```bash
sudo ./audio
```
However the behavior is not the same as if you run it without `sudo`, so the solution was to run it just like:
```bash
./audio
```
The reason why I was running with sudo was because I'm also accessing to the GPIO in the same program, which requires root privileges

closed 2023-09-06 by jsgalarraga

larsimmisch, 2023-09-06:
Glad that you figured it out. If you need to run your program as root, maybe you need to add `root` to the `audio` group.

jsgalarraga, 2023-09-06:
I ended up changing the program ownership and permissions, and worked great
```bash
sudo chown root:root audio
sudo chmod 4755 audio
./audio
```