larsimmisch/pyalsaaudio #139: Feature request: Expose non-interleaved write

oomzay, 2024-01-19:
For some applications non-interleaved channel i/o (snd_pcm_writen) is more convenient, and likely more efficient than snd_pcm_writei.

larsimmisch, 2024-01-20:
Just implementing `snd_pcm_writen` should be fairly simple, but updating the documentation would be a chore for me.

RonaldAJ, 2024-01-25:
@oomzay Can you draft documentation for the function you would like to have?

oomzay, 2024-01-25:
PCM.write_noninterleaved(data_n)
Writes (plays) the sound in data_n. Data_n must be a list of sample buffers, one buffer per output channel. The length of each buffer must be a multiple of the sample size, and should be exactly the size of a period. Every buffer must hold the same number of samples. If less than ‘period size’ frames are provided, the actual playout will not happen until more data is written.
(Then as for PCM.write())---John ChildMob: +44 (0)7879 828486

oomzay, 2024-01-25:
PCM.write_noninterleaved(data_n)

Writes (plays) the sound in data_n. Data_n must be a list of sample buffers, one buffer per output channel. The length of each buffer must be a multiple of the sample size, and should be exactly the size of a period. Every buffer must hold the same number of samples. If less than ‘period size’ frames are provided, the actual playout will not happen until more data is written.

(Then as for PCM.write())